okadan/flutter-nfc-manager #3: Exception: decodeObjectForKey: class "NFFieldNotification" not loaded or does not exist

GauravBapat93, 2019-11-20:
IOS version: 13.2.3

Error occurs after initiation of Read/ Write events on both devices.

> [xpc.exceptions] <NSXPCConnection:> connection to service on pid 85 named com.apple.nfcd.service.corenfc: Exception caught during decoding of received selector didDetectExternalReaderWithNotification:, dropping incoming message.

> Exception: Exception while decoding argument 0 (#2 of invocation):
> Exception: decodeObjectForKey: class "NFFieldNotificationECP1_0" not loaded or does not exist

Note: Events are getting triggered, Ready To scan Popup is disappearing.

okadan, 2019-11-21:
@1993-GB 

I think this is an issue related to iOS, not this plugin.

Anyway, I need more information.

- Provide your `flutter doctor -v`
- Make sure that you follow the setup instruction of `README.md`

1993-GB, 2019-11-22:
@okadan 

I am running Example app as well.. it is giving same errors.

Here is Flutter doctor -v result

[✓] Flutter (Channel stable, v1.9.1+hotfix.6, on Mac OS X 10.14.6 18G1012, locale en-JP)
    • Flutter version 1.9.1+hotfix.6 at /Users/XXX/Documents/flutter
    • Framework revision 68587a0916 (10 weeks ago), 2019-09-13 19:46:58 -0700
    • Engine revision b863200c37
    • Dart version 2.5.0

[!] Android toolchain - develop for Android devices (Android SDK version 28.0.3)
    • Android SDK at /Users/XXX/Library/Android/sdk
    • Android NDK location not configured (optional; useful for native profiling support)
    • Platform android-29, build-tools 28.0.3
    • Java binary at: /Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b49-5587405)
    ! Some Android licenses not accepted.  To resolve this, run: flutter doctor --android-licenses

[✓] Xcode - develop for iOS and macOS (Xcode 11.2.1)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 11.2.1, Build version 11B500
    • CocoaPods version 1.8.4

[✓] Android Studio (version 3.5)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin version 41.0.2
    • Dart plugin version 191.8593
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b49-5587405)

[!] IntelliJ IDEA Community Edition (version 2018.3.5)
    • IntelliJ at /Applications/IntelliJ IDEA CE.app
    ✗ Flutter plugin not installed; this adds Flutter specific functionality.
    ✗ Dart plugin not installed; this adds Dart specific functionality.
    • For information about installing plugins, see
      https://flutter.dev/intellij-setup/#installing-the-plugins

[✓] Connected device (1 available)
    • iPhone • XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX • ios • iOS 13.2.3

okadan, 2019-11-22:
@1993-GB 

Thanks.

Are `Provisioning Profile` and `Runner.entitlements` appropriate?

1993-GB, 2019-11-25:
@okadan 

Yes.. I have a new developer account and i have completed all steps in the Readme file.

okadan, 2019-11-25:
@1993-GB 

Please check the result of `NfcManager.isAvailable(NfcSessionType.tag)`.

1993-GB, 2019-11-25:
@okadan 

is there any minimum handset version requirement for iphone?

Which device are you testing on ..?

Because Apple Reference app is also not working [Apple Reference app](https://developer.apple.com/documentation/corenfc/building_an_nfc_tag-reader_app)

It is also exiting with the same error.

okadan, 2019-11-25:
> is there any minimum handset version requirement for iphone?

To use `startTagSession`, iOS13 or later is required.
@1993-GB 

> Which device are you testing on ..?

iOS13.2.3 and so on.

I still think this is an iOS layer issue, not a plugin issue.

Check `NfcManager.isAvailable` to make sure that your iOS setup is valid.

1993-GB, 2019-11-25:
@okadan 

Both Tag and Ndef are returning True for isAvailable!!

okadan, 2019-11-25:
@1993-GB 

Sorry, I have no ideas.

> Because Apple Reference app is also not working Apple Reference app

Does this mean that the same error occurs in a native iOS project?

In that case, it's not an issue with this plugin.

1993-GB, 2019-11-25:
@okadan 

Yes !! it looks like so!
Sorry for the trouble caused...

closed 2019-11-25 by 1993-GB

1993-GB, 2019-12-05:
@okadan 

Can we use this framework for communication between two iPhones?

jams032, 2022-06-15:
@1993-tensa-g , I have the same question. But i think the answer is YES. Recently, we were capable to writing data in Android phone and read that from iOS app. The code was based on java and iOS project is based on swift. The sad part is, we still are not capable of writing data in iOS, and read from Android app as well. This is the challenge we are struggling.