okadan/flutter-nfc-manager #7: isoDep transceive exception

adminant, 2020-03-03:
I try to get Visa Cardholder name, so I must call transceiver a couple of times, transmitting control codes, and in startTagSession I do:

var n = IsoDep.fromTag(tag);
Uint8List res1 = await n.transceive(SELECT);
print('res1: ' + res1.toString());
Uint8List res2 = await n.transceive(GETGPO);
print('res1: ' + res2.toString());

first transceive completes ok, but on second transceive I get:
----
Unhandled Exception: PlatformException(not_found, Tag is not found., null)
----
What am I doing wrong?

okadan, 2020-03-03:
@adminant 

Thank you for reporting an issue. I think this is a bug.

We probably need to await `_onTagDiscovered` on [L116](https://github.com/okadan/flutter_nfc_manager/blob/master/lib/src/nfc_manager/nfc_manager.dart#L116) (and also to change typedef on [L12](https://github.com/okadan/flutter_nfc_manager/blob/master/lib/src/nfc_manager/nfc_manager.dart#L12))

okadan, 2020-03-03:
@adminant  Fixed this bug and published v1.2.0 to pub.dev.

okadan, 2020-03-03:
@adminant 

Fixed this bug and published v1.2.0 to pub.dev.

Please let me know if the issue still exists.

adminant, 2020-03-04:
Yes, I tested, now it works, thank you!

closed 2020-03-06 by okadan