okadan/flutter-nfc-manager #8: iOS popover cancel button event

eddiesTime, 2020-03-06:
Hi @okadan 

first of all great work!

I am using your package in a project where I read a NfcTag and later use the tag id to call some server logic. Overall I am using the BLoC-pattern and I am wondering if there is an option to catch the "on cancel"-event when you tap the button on the Nfc-Popover in iOS. I differentiate between two states `IsReadyToReadNfc` and `ReadingNfc`. Depending on the state I have a button and want to display either "Start Scanning" or "Scanning..."

okadan, 2020-03-06:
Hi @eddiesTime 

Currently, this plugin does not support handling "cancel / error" events.

okadan, 2020-03-06:
Hi @eddiesTime

Currently, this plugin does not support catching "cancel / error" events.

But we can probably do this by handling [didInvalidateWithError](https://developer.apple.com/documentation/corenfc/nfctagreadersessiondelegate/3282001-tagreadersession) on iOS and specifying the callback when starting a session.

``` dart
NfcManager.instance.startTagSession(
  didInvalidateWithErrorIOS: (e) {
    // handling error
  },
);
```

eddiesTime, 2020-03-06:
Hi @okadan 

thank you for your fast response!

I can't wait to see what your next update will bring. In my opinion your plugin is the easiest to use and will be the best one at pub.dev yet with the addition of error handling.

okadan, 2020-03-17:
@eddiesTime 

Added a callback parameter for handling errors from the session. #10

And, published to pub.dev as v1.3.0.

eddiesTime, 2020-03-18:
Hi @okadan 

Excuse me for my delayed response. Thank you for updating the package with the new enhancement.

I've tried it out today but unfortunately did not receive any result. Do you mind to take a look at my approach and give me some feedback?

Down below is my implementation of the `startTagSession` and my approach to test the `onSessionError`-Callback.

```
NfcManager.instance.startTagSession(
        onDiscovered: (NfcTag tag) async {
      // discover logic
    }, onSessionError: (NfcSessionError error) async {
      print('error');
      print(error.message);
     // error logic
    });
```
When I hit the "Cancel"-Button on iOS nothing happens. I don't receive any print statements in my debug console.

I've taken a loot in your adaptation of nfc_manager.dart. The callback should respond with a NfcSessionError with the NfcSessionErrorType userCanceled when I select the cancel button.

okadan, 2020-03-18:
@eddiesTime

Sorry, this is a bug.

Fixed and published to pub.dev as v1.3.1.

eddiesTime, 2020-03-19:
Thank you.

I've tested it and it works as intended. Keep the good work up! :)

Closing the issue now.

closed 2020-03-19 by eddiesTime