okadan/flutter-nfc-manager #9: Read data from the sector?

speedlightwp, 2020-03-06:
Hi,

I am able to see the information like "id", "nfca", "maxTransceiveLength", "mifareclassic". Under "mifareclassic" there is block count, maxtransceivelength, sectorcount, size and timeout information. 

Are we able to read the data from different sector? I wish to retrieve the information in the card.

Thanks!

okadan, 2020-03-07:
Hi.

I'm considering creating a wrapper class for [MifareClassic](https://developer.android.com/reference/android/nfc/tech/MifareClassic) in the future.

At the moment, you can probably do that with `NfcA#transceive`.

speedlightwp, 2020-03-07:
Thanks @okadan!

I am new to NFC. I am testing on isodep card too. Sending the APDU header "00A40400". I received the following:

[111, 101, 132, 8, 160, 0, 0, 0, 24, 67, 77, 0, 165, 89, 115, 74, 6, 7, 42, 134, 72, 134, 252, 107, 1, 96, 12, 6, 10, 42, 134, 72, 134, 252, 107, 2, 2, 1, 1, 99, 9, 6, 7, 42, 134, 72, 134, 252, 107, 3, 100, 11, 6, 9, 42, 134, 72, 134, 252, 107, 4, 2, 85, 101, 11, 6, 9, 43, 133, 16, 134, 72, 100, 2, 1, 3, 102, 12, 6, 10, 43, 6, 1, 4, 1, 42, 2, 110, 1, 2, 159, 110, 6, 25, 129, 34, 2, 1, 0, 159, 101, 1, 255, 144, 0]

The response data is 0x9000 (144,0) is success however I do not know how to interpret the rest of the data and decode the payload to get any meaningful information out of it. Do you know of any links I can read up to decode or to get more information out of the card?

Thanks!

okadan, 2020-03-08:
@speedlightwp 

Sorry, I don't know anything other than those published on the web...

Please explore yourself.

smlu, 2020-03-09:
> Thanks @okadan!
>
> I am new to NFC. I am testing on isodep card too. Sending the APDU header "00A40400". I received the following:
>
> [111, 101, 132, 8, 160, 0, 0, 0, 24, 67, 77, 0, 165, 89, 115, 74, 6, [...]

@speedlightwp some information on what's being returned by ICC can be found in ISO/IEC 7816-4 sepcification.
Looking at section [5.1.5](https://cardwerk.com/smart-card-standard-iso7816-4-section-5-basic-organizations/?elementor-preview&1514313288596#chap5_1_5) of ISO/IEC 7816-4 doc I would say above bytes represents FCI template (`0x6F`). 
More information about `SELECT FILE` command you can find [here](https://cardwerk.com/smart-card-standard-iso7816-4-section-6-basic-interindustry-commands/#chap6_11)

okadan, 2020-04-20:
Closing because there is no response.

closed 2020-04-20 by okadan