okadan/flutter-nfc-manager #11: iOS ISO15693 read Multiple Blocks

Lastone17, 2020-03-26:
I always get "Invalid Parameters" & "custom Command code must be between 0xA0 to 0xDF"

What I do:
`        Uint8List command = new Uint8List(10);
        command[0] = (182);
        command[1] = (51);
        command[2] = (59);
        command[3] = (167);
        command[4] = (0);
        command[5] = (38);
        command[6] = (2);
        command[7] = (224);
        command[8] = (165);
        command[9] = (4);
        Uint8List response = await iso15693.customCommand(requestFlags: requestFlags, commandCode: 23, parameters: command);
        debugPrint('Response: ' +  response.toString());`

equals to the command: 22 23 B6 33 3B A7 00 26 02 E0 A5 04
right?

But I can't get any response.

okadan, 2020-03-26:
@Lastone17 

This wraps `NFCISO15693Tag#customCommand` API on iOS.

See the [doc](https://developer.apple.com/documentation/corenfc/nfciso15693tag/3043799-customcommand).

> Sends a custom command (0xA0 to 0xDF command code), as defined in the ISO 15693-3 specification, to the tag.

So we need to create a wrapper for each of the commands such as [readMultipleBlocks](https://developer.apple.com/documentation/corenfc/nfciso15693tag/3043809-readmultipleblocks).

Lastone17, 2020-03-26:
Im pretty new to flutter, I hoped it's possible to use the command readMultipleBlocks with flutter on iOS.
Can you help me with this warpper for 'readMultipleBlocks'? Or do you already implemented such?

Lastone17, 2020-03-26:
> Im pretty new to flutter, I hoped it's possible to use the command readMultipleBlocks with flutter on iOS.
> Can you help me with this warpper for 'readMultipleBlocks'? Or do you already implemented such?

okadan, 2020-03-26:
@Lastone17 

Currently not implemented, but I will do this in the future.

Lastone17, 2020-03-26:
So no other way to use readmultipleBlocks with flutter on iOS?
Can I help you somehow?
I dont want to presure, but I really need this command this or next week.

okadan, 2020-03-26:
@Lastone17 

> So no other way to use readmultipleBlocks with flutter on iOS?

okadan, 2020-03-26:
@Lastone17 

> So no other way to use readmultipleBlocks with flutter on iOS?

The only way is to implement your own [platform-channel](https://flutter.dev/docs/development/platform-integration/platform-channels).

> I dont want to presure, but I really need this command this or next week.

OK I will try it on weekend.

Lastone17, 2020-03-26:
@okadan 
> I will try it on weekend.
Wow that would be great!

okadan, 2020-03-28:
@Lastone17 

I started #12 .

1.4.0-dev version that includes the ISO15693 commands implementation is now available on pub.dev.

Lastone17, 2020-03-28:
@okadan 
Very cool, thanks! I will test this and inform you about it.

Lastone17, 2020-03-30:
@okadan 

So I think i found several bugs:
Unhandled Exception: type 'List<dynamic>' is not a subtype of type 'List<Uint8List>'
`Future<List<Uint8List>> readMultipleBlocks({`

should be:
`Future<Uint8List> readMultipleBlocks({`

Because we want a Uint8List not a List<Uint8List> right? 

Als I tried this command:
I wanted to read from A05h and 4 blocks of numbers:
`        Set<ISO15693RequestFlag> requestFlags = new Set();
        requestFlags.add(ISO15693RequestFlag.highDataRate);
        requestFlags.add(ISO15693RequestFlag.address);
        Uint8List response = await iso15693.extendedReadMultipleBlocks(requestFlags: requestFlags, blockNumber: 165, numberOfBlocks: 4);`

I get this error from Core NFC of iOS:
Block range must be between 0 to 255 inclusively
Unhandled Exception: PlatformException(error, Parameter value is out of bound, null)

next I tried 'extendedReadMultipleBlocks' starting block 0 and also 4 blocks of number:
Unhandled Exception: PlatformException
StandardMethodCodec.decodeEnvelope

I got this error from Core NFC of iOS.

Thanks for your help for these 3 errors.

Lastone17, 2020-03-30:
@okadan 
Update:
I also tested the 'readSingleBlock' - this works like it should.
`        Set<ISO15693RequestFlag> requestFlags = new Set();
        requestFlags.add(ISO15693RequestFlag.highDataRate);
        requestFlags.add(ISO15693RequestFlag.address);
Uint8List response = await iso15693.readSingleBlock(requestFlags: requestFlags, blockNumber: 165);`

ReadMultipleBlocks needs some correction, as I mentioned before.
Thanks for your work!

okadan, 2020-03-30:
@Lastone17 

Thank you for your report.

I will fix bugs and APIs based on feedback and testing, and publish v1.4.0-dev to pub.dev continuously.

Lastone17, 2020-03-30:
@okadan 

Thanks I will test further as soon as the fixed version of readMultipleBlocks is released.

okadan, 2020-03-30:
@Lastone17 

v1.4.0-dev+2 with the fix for above bugs is now available.

It is still a development version and has not been fully tested. Sorry.

Lastone17, 2020-03-30:
@okadan 
No problem, I'm glad I can help you with testing of these commands.

I now tested readMultipleBlocks - works like it have to!

About extendedReadMultipleBlocks from iOS (NFC Chip Manufacturer - STMicroelectronics):
"Basically, as per ISO/IEC 15693-3, the "First block number" parameter of Extended read multiple blocks command is a 2-bytes parameter whereas iOS13 encodes it as a 1-byte parameter causing the command to be erroneous. Therefore, any tag properly supporting the ISO 15693-3 Extended Read Multiple Blocks command will reject the extendedReadMultipleBlocks sent by iOS 13 as being erroneous."

So we can test this if Apple iOS fixes the issue.

I will test even further your plugin while developing my ISO 15693 NFC App

Thanks for your amazing and fast work!
P.s If you wish you can keep in track of the NFC App we are developing and of course using your plugin.
Contact me if you want - dominik.muggli@muggli-designs.ch

closed 2020-03-30 by Lastone17