okadan/flutter-nfc-manager #16: Failed to load CoreNFC on ios crash.

sgehrman, 2020-04-30:
dyld: Library not loaded: /System/Library/Frameworks/CoreNFC.framework/CoreNFC
  Referenced from: /var/containers/Bundle/Application/EB1AFB0C-E3D8-4400-ACED-58AB61EA86F9/Runner.app/Runner
  Reason: image not found

Possible fix:  https://stackoverflow.com/questions/44946057/dyld-library-not-loaded-rpath-corenfc-framework-corenfc-ios11-and-xcode-9-be

Could you add this: 
#if canImport(CoreNFC)

to the code? 

I just updated Xcode, not sure if this was the cause, but it just started happening.

AirborneEagle, 2020-06-05:
Are you saying you were able to get it working on the iPhone 6, but it has since stopped working?
From what I can tell, NFC reads were not allowed by third party applications on anything prior to the iPhone 7.
https://www.bluebite.com/nfc/iphone-nfc-compatibility
I have a 6s, and I am finding that `await NfcManager.instance.isAvailable()` returns `false`
and if I try to read a card anyways, nothing happens. 

pbspec: ` nfc_manager: ^1.3.2+4`

ThirupathiRaoM, 2022-01-24:
Hello Every one 
I am Working coreNFC Project I can Able read and write the NFC Tag I am Getting issue while password protected the NFC device using CoreNFC if any body having the idea regarding please let us know it will help me a lot 
Thanks & Regards
iOS Developer

closed 2022-07-20 by okadan