okadan/flutter-nfc-manager #17: QUESTION: Reading previous write Tag Data

alexlovar, 2020-05-13:
Hi, i like to learn about nfc, i use write tag with your sample and get this response:
Success to "Ndef Write"

but when i read the tag i get this:

{id: [249, 109, 254, 163], nfca: {atqa: [4, 0], maxTransceiveLength: 253, sak: 8, timeout: 618}, mifareclassic: {blockCount: 64, maxTransceiveLength: 253, sectorCount: 16, size: 1024, timeout: 618, type: 0}, ndef: {cachedMessage: {records: [{typeNameFormat: 1, type: [84], identifier: [], payload: [2, 101, 110, 72, 101, 108, 108, 111, 32, 87, 111, 114, 108, 100, 33]}, {typeNameFormat: 1, type: [85], identifier: [], payload: [4, 102, 108, 117, 116, 116, 101, 114, 46, 100, 101, 118]}, {typeNameFormat: 2, type: [116, 101, 120, 116, 47, 112, 108, 97, 105, 110], identifier: [], payload: [72, 101, 108, 108, 111]}, {typeNameFormat: 4, type: [99, 111, 109, 46, 101, 120, 97, 109, 112, 108, 101, 58, 109, 121, 116, 121, 112, 101], identifier: [], payload: [109, 121, 100, 97, 116, 97]}]}, canMakeReadOnly: false, isWritable: true, maxSize: 716, type: com.nxp.ndef.mifareclassic}}

how i do to process the payload to see the info as was wrote?

thanks for your help!

okadan, 2020-05-14:
You need to parse it by yourself based on the NDEF specifications.

For example,

``` dart
final NdefRecord record = someGetNdefRecordFunction();

if (record.typeNameFormat == 0x01 && record.type.length == 1 && record.type.first == 0x54) {
  // This record is `NFC Wellknown Text`.
  final int languageCodeLength = ndefRecord.payload.first;
  final String languageCode = ascii.decode(ndefRecord.payload.sublist(1, 1 + languageCodeLength));
  final String text = utf8.decode(ndefRecord.payload.sublist(1 + languageCodeLength));
} else if ...
```

ibcoleman, 2021-03-29:
Anyone? The API seems to support getting NdefRecords but where's the support for getting useful values out of that record?

BrayFlex, 2021-04-13:
Map tagData = tag.data;
Map tagNdef = tagData['ndef'];
Map cachedMessage = tagNdef['cachedMessage'];
Map records = cachedMessage['records'][0];
Uint8List payload = records['payload'];
String payloadAsString = String.fromCharCodes(payload);

okadan, 2021-04-14:
```dart
```

BrayFlex, 2021-04-14:
After trying 4 different libraries I would recommend anyone who is targeting min sdk 28+ to fork and use this repo: https://github.com/vilvic/nfc_in_flutter save yourself the headaches

okadan, 2021-04-14:
@BrayFlex You should not access the `tag.data` property directly. Use `Ndef.from` to obtain an `Ndef` instance.

```dart
final ndef = Ndef.from(tag);
if (ndef == null) throw('Tag is not ndef.');
print(String.fromCharCodes(ndef.cachedMessage.records[0].payload));
```

@ibcoleman 
> The API seems to support getting NdefRecords but where's the support for getting useful values out of that record?

This plugin is just a bridge to the Android/iOS APIs, and does not do what these APIs do not.

So you need to decode `Uint8List` based on the NDEF specification. For example:

```dart
final record = ndef.cachedMessage.records[0];

if (record.typeNameFormat == NdefTypeNameFormat.nfcWellknown && record.type.length == 1 && record.type.first == 0x54) {
  // record type: NFC Wellknown Text
  final languageCodeLength = record.payload.first;
  final languageCode = ascii.decode(record.payload.sublist(1, 1 + languageCodeLength));
  final text = utf8.decode(record.payload.sublist(1 + languageCodeLength));
} else if (...) // For more record types...
```

closed 2021-05-22 by okadan