okadan/flutter-nfc-manager #23: MetaData UUID

yunusemrecetin, 2020-06-02:
Can I access uuid of nfc? I can't find any information in your library code API and docs.

okadan, 2020-06-02:
You can access to:

- `FeliCa#currentIDm`
- `ISO7816#identifier`
- `ISO15693#identifier`
- `IsoDep#identifier`
- `MiFare#identifier`
- `NfcA#identifier`
- `NfcB#identifier`
- `NfcF#identifier`
- `NfcV#identifier`

yunusemrecetin, 2020-06-02:
Thanks for your quick answer. I know this library is only a bridge, I just want to get UUID of NFC tag. And thanks for this library and open-source your contributions.

`NfcManager.instance.startTagSession(onDiscovered: (NfcTag tag) async {
      tag.data.forEach((key, value) {
        print("key : " + key + " - value : " + value.toString());
        switch (key) {
          case "id":
            String id = String.fromCharCodes(value);
            print("id : " + id);
            break;
        }
      });
    });`

Console log;
`I/flutter (18004): key : id - value : [4, 142, 192, 130, 100, 103, 128]
I/flutter (18004): id : Àdg
I/flutter (18004): key : nfca - value : {atqa: [68, 0], maxTransceiveLength: 253, sak: 0, timeout: 618}
I/flutter (18004): key : mifareultralight - value : {maxTransceiveLength: 253, timeout: 618, type: 2}
I/flutter (18004): key : ndef - value : {cachedMessage: {records: [{typeNameFormat: 1, type: [85], identifier: [], payload: [4, 121, 117, 110, 117, 115, 101, 109, 114, 101, 99, 101, 116, 105, 110, 46, 99, 111, 109, 46, 116, 114]}]}, canMakeReadOnly: true, isWritable: true, maxSize: 137, type: org.nfcforum.ndef.type2}`

I tried these code block but I can't access the id of NFC tag.

Can you help me?
Thanks for everything.

okadan, 2020-06-02:
For example, to access NfcA properties on Android:

``` dart
// onDiscovered Callback
final NfcA nfcA = NfcA.fromTag(tag);
if (nfcA == null) {
  print('Tag is not NfcA');
  return;
}
print(nfcA.identifier);
```

closed 2020-06-02 by yunusemrecetin

yunusemrecetin, 2020-06-02:
It's returned : "[4, 142, 192, 130, 100, 103, 128]"
When I convert to string with String.fromCharCodes, the result is "Àdg"

Am I in the wrong way?

okadan, 2020-06-02:
Sorry, what is the value do you expect?

Anyway, it's not an issue related to this library, so please refer to the `Uint8List` reference or Stackoverflow.