okadan/flutter-nfc-manager #24: NFC session on Android fails to resume

markus-oehme-pg40-com, 2020-06-05:
I tried to release and reacquire an NFC session with the following code. This is so that I would get discovery callbacks after my app was intermittently paused (e.g. user switched to another app). The following uses the WidgetsBindingObserver to detect lifecycle changes.

<pre>
  @override
  void didChangeAppLifecycleState(AppLifecycleState state) {
    if (state == AppLifecycleState.resumed) {
      NfcManager.instance.startTagSession(
          onDiscovered: (NfcTag tag) async {
            handleNFC(tag);
          });
    } else {
      NfcManager.instance.stopSession();
    }
  }
</pre>

However after switching to another app and back my app does not get any more discovery callbacks. So discovery works from the start of the app until the first pause event.

alestiago, 2021-05-31:
I'm getting a similar issue, just when disposing another camera plugin the session does not get discovery callbacks. I'm currently looking into this issue I'm getting. It might be related to your problem.

markus-oehme-pg40, 2021-05-31:
We chose a different approach than flutter, so I don't have the code handy. The only thing (besides a bare app skeleton) necessary should be a `handleNFC` function doing some logging so the effect can be observed.