okadan/flutter-nfc-manager #25: NFC transceive fails at second transmission on Android.

markus-oehme-pg40-com, 2020-06-05:
I try to use the NfcV class to transmit some commands to a corresponding tag. I registered the following function as callback for the onDiscovered events. The result is a correct answer on the first try and an exception on the second try. Did I do it wrong?

<pre>
Future<void> callbackFunction(NfcTag tag) async {
  final nfcVtag = NfcV.fromTag(tag);
  final cmd = [0x02, 0x20, 0x00].toList();
  try {
    final answer = await nfcVtag.transceive(Uint8List.fromList(cmd));
    debugPrint('First answer ' + answer.toString());
  } catch (e) {
    debugPrint('Exception on first try: ' + e.toString());
  }
  try {
    final answer = await nfcVtag.transceive(Uint8List.fromList(cmd));
    debugPrint('Second answer ' + answer.toString());
  } catch (e) {
    debugPrint('Exception on second try: ' + e.toString());
  }
}
</pre>

Output:
<pre>
I/flutter (17066): First answer [0, 167, 19, 240, 84, 5, 0, 3, 101]
I/flutter (17066): Exception on second try: PlatformException(not_found, Tag is not found., null)
</pre>

ishwaryamamidi, 2020-09-04:
@markus-oehme-pg40 I have the same issue. Did you find any solution to this already?

markus-oehme-pg40, 2020-09-04:
@ishwaryamamidi Sadly not. We ended up not using NFC on flutter.

okadan, 2020-09-04:
I will be investigating this issue in the near future.

ishwaryamamidi, 2020-09-04:
@okadan It would be very grateful if you could investigate this as soon as possible. As this is a major drawback currently.

okadan, 2020-09-04:
@ishwaryamamidi @markus-oehme-pg40 

I have found success under the following environments:

- Plugin Version: `nfc_manager: 2.0.0`
- Tag Type: I-CODE SLI (ISO15693 compatible)

``` dart
final nfcv = NfcV.from(tag);
final cmd = [0x02, 0x20, 0x00];

for (int i = 1; i <= 100; i += 1) {
  try {
    final answer = await nfcv.transceive(data: Uint8List.fromList(cmd));
    print('Answer $i: $answer');
  } catch (e) {
    print('Exception $i: $e');
  }
}
```

ishwaryamamidi, 2020-09-04:
@okadan Thank you for your quick investigation.
Even after updating to nfc_manager: 2.0.0, I get an error for second transceive.
Below is a sample of my code:

  Future<String> handleTag(NfcTag tag, Iterable<Record> recordList) async {
    final tech = NfcV.from(tag);
    if (tech == null) throw ('Tag is not compatible with NDEF.');
    /*if (!tech.isWritable)
      throw('Tag is not NDEF writable.');*/
    Uint8List mbtx = Uint8List.fromList([0xDA, 0x7A]);
    test_nfc(tag, mbtx);
  }

  test_nfc(NfcTag tag, Uint8List mbtx) async {
    //NfcTag temptag = tag;
    int tries = 0;
    int msglength;
    int mb_control_dyn = 0;
    await write_MB(tag, mbtx);
    tries = 0;
    while ((mb_control_dyn & 0x40) != 0x40) {
      tries++;
      if (tries > 5) {
        throw ('error in tries');
      }
      sleep(const Duration(milliseconds: 10));
      mb_control_dyn = await read_MB_control_dyn(tag);
    }
    sleep(const Duration(milliseconds: 10));
    msglength = await read_MB_length(tag);
    return await read_MB(tag, msglength);
  }

  Future<Uint8List> write_MB(NfcTag tag, Uint8List data) async {
    final tech = NfcV.from(tag);
    if (tech == null) throw ('Tag is not compatible with NDEF.');
    int i1 = 0x22;
    int i2 = 0xAA;
    int i3 = 0x02;
    int i11 = data.length - 1;
    Uint8List write_rx;
    List<int> write_temp = new List();
    transceive_id = tech.identifier;
    if ((data.length < 2) || (data.length > 255)) {
      throw ('new exception');
    }
    write_temp.insert(0, i1);
    write_temp.insert(1, i2);
    write_temp.insert(2, i3);
    write_temp.insertAll(3, transceive_id);
    write_temp.insert(11, i11);
    write_temp.insertAll(12, data);
    print('dummy $data');

    Uint8List wrdata = Uint8List.fromList(write_temp);
    print('this is write data $wrdata');
    write_rx = await my_nfctransceive(tag, wrdata);
    print('this is after trasceive of write ${tag.data}');
    print('this is write response $write_rx');
    //return write_rx;
  }

  Future<int> read_MB_control_dyn(NfcTag tag) async {
    final tech = NfcV.from(tag);
    if (tech == null) throw ('Tag is not compatible with NDEF.');
    int i1 = 0x22;
    int i2 = 0xAD;
    int i3 = 0x02;
    int i11 = 0x0D;
    List<int> temp = new List();
    Uint8List rxdata;
    //int response;
    transceive_id = tech.identifier;
    temp.insert(0, i1);
    temp.insert(1, i2);
    temp.insert(2, i3);
    temp.insertAll(3, transceive_id);
    temp.insert(11, i11);
    Uint8List txdata = Uint8List.fromList(temp);
    print('this is read_MB_control_dyc_temp $txdata');
    rxdata = await my_nfctransceive(tag, txdata);
    print('this is after trasceive of read_control ${tag.data}');
    print('this is read_MB_control_dyc $rxdata');
    if (rxdata[0] != 0) {
      print('Tag answered, but with error in Response_flags ');
    }

    print('this is read_MB_control_dyn ${rxdata[1]}');
    return rxdata[1];
  }

  Future<int> read_MB_length(NfcTag tag) async {
    print('length function');
    final tech = NfcV.from(tag);
    List<int> read_length_temp = new List();
    Uint8List read_MB_length_response;
    int i1 = 0x22;
    int i2 = 0xAB;
    int i3 = 0x02;
    transceive_id = tech.identifier;
    read_length_temp.insert(0, i1);
    read_length_temp.insert(1, i2);
    read_length_temp.insert(2, i3);
    read_length_temp.insertAll(3, transceive_id);
    Uint8List read_length = Uint8List.fromList(read_length_temp);
    print('length data $read_length');
    read_MB_length_response = await my_nfctransceive(tag, read_length);
    print('this is after trasceive of length ${tag.data}');
    print('this is read length response $read_MB_length_response');
    if (read_MB_length_response[0] != 0) {
      print('error on read MB length');
    }
    return read_MB_length_response[1] + 1;
  }

  Future<Uint8List> read_MB(NfcTag tag, int length) async {
    final tech = NfcV.from(tag);
    List<int> read_mb_temp = new List();
    Uint8List read_mb_response;
    int i1 = 0x22;
    int i2 = 0xAC;
    int i3 = 0x02;
    int i11 = 0x00;
    read_mb_temp.insert(0, i1);
    read_mb_temp.insert(1, i2);
    read_mb_temp.insert(2, i3);
    read_mb_temp.insertAll(3, transceive_id);
    read_mb_temp.insert(11, i11);
    read_mb_temp.insert(12, length - 1);
    Uint8List read_mb = Uint8List.fromList(read_mb_temp);
    read_mb_response = await my_nfctransceive(tag, read_mb);
    print('this is after trasceive of read ${tag.data}');
    if ((read_mb_response[0] != 0) ||
        (read_mb_response.length !=
            length +
                1)) /* Tag answered a different length (response byte + data length)  */
    {
      print('error in mb_read');
    }
    Uint8List retval = read_mb_response.sublist(1);
    print('this is retval $retval');
    return retval;
  }

  my_nfctransceive(NfcTag tag, Uint8List my_tx) async {
    print('hello1');
    final tech = NfcV.from(tag);
    int tries = 0;
    Uint8List retval;
    if (tag == null) {
      print("No TagV !");
      return null;
    }
    while (true) {
      try {
        tries++;
        print(tries);
        retval = await tech.transceive(data: my_tx);
        print('hello2');
        break;
      } catch (e) {
        if (tries >= 5) {
          throw (e);
        }
        // throw (e);
        print('hello3');
        sleep(const Duration(milliseconds: 10));
      }
    }
    print('this is my_transceive $retval');
    return retval;
  }
}

Could you help me out with this? Thanks in advance. :)

okadan, 2020-09-04:
@ishwaryamamidi 

Please try with the minimal code like bellow:

``` dart
final nfcv = NfcV.from(tag);
final cmd = [0x02, 0x20, 0x00];

final n = 2;

for (int i = 1; i <= n; i += 1) {
  try {
    final answer = await nfcv.transceive(data: Uint8List.fromList(cmd));
    print('Answer $i: $answer');
  } catch (e) {
    print('Exception $i: $e');
  }
}
```

ishwaryamamidi, 2020-09-04:
@okadan The above minimal code worked perfectly. 
But could you please let me know what is the issue in the code I have written.

okadan, 2020-12-02:
Close issue because the second transmission was successful.

closed 2020-12-02 by okadan