okadan/flutter-nfc-manager #27: iOS ISO7816Tag.sendCommand returns empty string

BryanBlack523, 2020-08-20:
After tag session started and tag is found, commands are sent down to card, but there are no response codes to see if it actually succeded.
Code snippet:
`NfcManager.instance.startTagSession(onDiscovered: (NfcTag tag) async {
    dynamic c;
    Platform.isAndroid ? c = IsoDep.fromTag(tag) : c = ISO7816.fromTag(tag);
    Uint8List aid = Uint8List.fromList(
              [0xd3, 0x92, 0xf0, 0x00, 0x26, 0x01, 0x00, 0x00, 0x00, 0x01]);
    String recvData;
   try{
          Uint8List recvDataByte;
          Platform.isAndroid
              ? recvDataByte = await c.transceive(commandLine)
              : recvDataByte = await c.sendCommand(
                  instructionClass: 0x00, //CLA
                  instructionCode: 0xA4, //INS
                  p1Parameter: 0x04, //P1
                  p2Parameter: 0x0C, //P2
                  data: aid,
                  expectedResponseLength: 0xFF); //Le

          while (recvDataByte == null) bool wait = true;
          recvData = bytesToString(recvDataByte);

          print("tested comand result: " + recvData);
        } catch (e) {
          print(e.toString());
        }
}`
Tested card in example is a MyNumber card.

okadan, 2020-08-20:
This may be due to the fact that it does not return statusWord1 and statusWord2 in ver 1.4.0-dev.

I plan to re-implement these commands in ver 2.0.0-dev.

okadan, 2020-09-10:
v2.0.0 has been published. It's not possible to get statusWord1 and statusWord2.