okadan/flutter-nfc-manager #28: Issue with non-NFC devices

selmaskopljakovic, 2020-09-23:
I made an android app that has NFC scanning and it works perfectly. For non-NFC devices, I made another option for scanning. The problem is that I can't even build an app on a non-NFC device. There is an exception:

```
Caused by: java.lang.IllegalStateException: NfcAdapter.getDefaultAda…nding.applicationContext) must not be null
E/AndroidRuntime(30997): 	at io.flutter.plugins.nfcmanager.NfcManagerPlugin.onAttachedToEngine(NfcManagerPlugin.kt:65)
E/AndroidRuntime(30997): 	at io.flutter.embedding.engine.FlutterEnginePluginReg

```

What to do to have buildable app for any android device but to use NFC with nfc-manager on device with NFC?
Thanks

okadan, 2020-09-25:
Thank you for reporting.

I think the plugin needs to be fixed.

I will publish a new version on pub.dev soon.

okadan, 2020-09-26:
@selmaskopljakovic 

Fixed and published v2.0.1 to pub.dev.

Commit: https://github.com/okadan/flutter-nfc-manager/commit/61e9af15d215a36f22ec8956eb1897e3d223b02d

selmaskopljakovic, 2020-09-26:
@okadan 
I just checked and it works :). 
Great!
Thank you

closed 2020-09-26 by okadan