okadan/flutter-nfc-manager #29: NfcManager plugin causes app to crash on launch

patuoynageek, 2020-10-04:
Not sure why this is happening?

`Installing build/app/outputs/flutter-apk/app.apk...
E/AndroidRuntime(22220): FATAL EXCEPTION: main
E/AndroidRuntime(22220): Process:com.app.name.developing, PID: 22220
E/AndroidRuntime(22220): java.lang.IllegalStateException: NfcAdapter.getDefaultAda…nding.applicationContext) must not be null
E/AndroidRuntime(22220): 	at io.flutter.plugins.nfcmanager.NfcManagerPlugin.onAttachedToEngine(NfcManagerPlugin.kt:65)
E/AndroidRuntime(22220): 	at io.flutter.embedding.engine.FlutterEnginePluginRegistry.add(FlutterEnginePluginRegistry.java:133)
E/AndroidRuntime(22220): 	at io.flutter.plugins.GeneratedPluginRegistrant.registerWith(GeneratedPluginRegistrant.java:51)
E/AndroidRuntime(22220): 	at com.app.name_flutter.MainActivity.configureFlutterEngine(MainActivity.kt:41)
E/AndroidRuntime(22220): 	at io.flutter.embedding.android.FlutterFragment.configureFlutterEngine(FlutterFragment.java:949)
E/AndroidRuntime(22220): 	at io.flutter.embedding.android.FlutterActivityAndFragmentDelegate.onAttach(FlutterActivityAndFragmentDelegate.java:180)
E/AndroidRuntime(22220): 	at io.flutter.embedding.android.FlutterFragment.onAttach(FlutterFragment.java:583)
E/AndroidRuntime(22220): 	at androidx.fragment.app.Fragment.performAttach(Fragment.java:2574)
E/AndroidRuntime(22220): 	at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:828)
E/AndroidRuntime(22220): 	at androidx.fragment.app.FragmentTransition.addToFirstInLastOut(FragmentTransition.java:1197)
E/AndroidRuntime(22220): 	at androidx.fragment.app.FragmentTransition.calculateFragments(FragmentTransition.java:1080)
E/AndroidRuntime(22220): 	at androidx.fragment.app.FragmentTransition.startTransitions(FragmentTransition.java:119)
E/AndroidRuntime(22220): 	at androidx.fragment.app.FragmentManagerImpl.executeOpsTogether(FragmentManagerImpl.java:1866)
E/AndroidRuntime(22220): 	at androidx.fragment.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManagerImpl.java:1824)
E/AndroidRuntime(22220): 	at androidx.fragment.app.FragmentManagerImpl.execPendingActions(FragmentManagerImpl.java:1727)
E/AndroidRuntime(22220): 	at androidx.fragment.app.FragmentManagerImpl.dispatchStateChange(FragmentManagerImpl.java:2663)
E/AndroidRuntime(22220): 	at androidx.fragment.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManagerImpl.java:2613)
E/AndroidRuntime(22220): 	at androidx.fragment.app.FragmentController.dispatchActivityCreated(FragmentController.java:246)
E/AndroidRuntime(22220): 	at androidx.fragment.app.FragmentActivity.onStart(FragmentActivity.java:542)
E/AndroidRuntime(22220): 	at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1419)
E/AndroidRuntime(22220): 	at android.app.Activity.performStart(Activity.java:7479)
E/AndroidRuntime(22220): 	at android.app.ActivityThread.handleStartActivity(ActivityThread.java:3454)
E/AndroidRuntime(22220): 	at android.app.servertransaction.TransactionExecutor.performLifecycleSequence(TransactionExecutor.java:180)
E/AndroidRuntime(22220): 	at android.app.servertransaction.TransactionExecutor.cycleToPath(TransactionExecutor.java:165)
E/AndroidRuntime(22220): 	at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:142)
E/AndroidRuntime(22220): 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:70)
E/AndroidRuntime(22220): 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2199)
E/AndroidRuntime(22220): 	at android.os.Handler.dispatchMessage(Handler.java:112)
E/AndroidRuntime(22220): 	at android.os.Looper.loop(Looper.java:216)
E/AndroidRuntime(22220): 	at android.app.ActivityThread.main(ActivityThread.java:7625)
E/AndroidRuntime(22220): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(22220): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:524)
E/AndroidRuntime(22220): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:987)`

okadan, 2020-10-04:
@patuoynageek

This is because the device does not support NFC.

Please upgrade plugin to v2.0.1.

patuoynageek, 2020-10-05:
Thanks, you're right, issue of app unable to launch on devices without NFC support fixed for v2.0.1

closed 2020-10-05 by patuoynageek