okadan/flutter-nfc-manager #31: Will there be any implementation of getting the UID/ID from an NFC Tag?

jarren-meep, 2020-10-29:
Hello, I'm trying to make a login app which utilises the UID/ID of an NFC Card/Tag to log the user in. I can obtain the value in Android using nfc_in_flutter (another package) but I cant use that here because it doesn't seem to even register the nfc card being infront of the phone. Only yours (nfc_manager) works and actually detects the card when placed infront of the phone.

The NFC tag I'm using currently is ISO14443, so I set the polling option to that within session start.

And so I ask, will there be any implementation of getting UID/ID from NFC tags? Because currently on session start of the NFC Manager, once we hover the nfc tag over, we get back 
`{mifare: {identifier: [4, 40, 252, 242, 37, 94, 129], historicalBytes: [], mifareFamily: 2}, ndef: {isWritable: true, maxSize: 137}}`
the data received seems to be everything *but* the ID, which is a bit weird.

Thanks!

okadan, 2020-10-29:
@jarren-meep 
For examle, to get ID from MiFare on iOS:

```dart
// onDiscovered callback
final mifare = MiFare.from(tag);
if (mifare == null) {
  print('Tag is not compatible with MiFare.');
  return;
}
print(mifare.identifier);
```

jarren-meep, 2020-10-29:
That gives me the Uint8list, which is what I have up there, as the identifier.
Do I decode it with UTF8 to get the actual ID? or what, because this isn't the value that I'm looking for. 
The NFC card has a code inside it that is '0248FCF2255E81', and I need to get this value.

okadan, 2020-10-29:
@jarren-meep 
To get a hexadecimal string such as `0248FCF2255E81`:

```dart
final String identifier = mifare.identifier.map((e) => e.toRadixString(16)).join('');
print(identifier); // => 428fcf2255e81
```

jarren-meep, 2020-10-30:
Just got to test it and it works, thank you @okadan !
am quite a newbie but now I understand how it gives back the identifier, thank uuu

closed 2020-10-30 by jarren-meep

Kurkkuo, 2022-05-11:
I was having the same issue, thank you @okadan!

firstmetcs, 2023-09-04:
> Is it possible to do get the UID from the Ndef tag?

+1