okadan/flutter-nfc-manager #38: ios 14.3 doesn not open nfc reader dialog

LtotheWT, 2021-01-18:
NfcManager.instance.startSession works in android but not ios

okadan, 2021-01-18:
Are you running on the real device instead of Simulator?

Have you checked the iOS Setup on README?

LtotheWT, 2021-01-18:
yeah i did, it causes other nfc app not able to open nfc too.

aleksey-smx, 2021-01-18:
the same issue. iphone 11 (ios 14.3)

iphone XR (ios 14.01 and 13.5.1) everything ok

LtotheWT, 2021-01-19:
https://developer.apple.com/forums/thread/657166 looks like its the NFCTagReaderSession problem

xleliberty, 2021-01-25:
Hi same goes here with iphone XS and ios 14.3
Any hint ?
thanks

ayyoubelamrani4, 2021-01-25:
Hello, same here for iPhone 7, iOS 14.3

okadan, 2021-01-25:
Hmm...

This seems to be an iOS system issue.

There seems to be nothing this plugin can do.

Do you have any other ideas?

I will investigate this issue near the future.

ayyoubelamrani4, 2021-01-25:
> Hmm...
> 
> 
> 
> This seems to be an iOS system issue.
> 
> 
> 
> There seems to be nothing this plugin can do.
> 
> 
> 
> Do you have any other ideas?
> 
> 
> 
> I will investigate this issue near the future.

It works on other apps using NFC readings perfectly. However, once I try opening an app with the flutter nFc manager package, the window inviting the user to « hold the phone near the tag» does not appear anymore. Right after this, the other apps where the NFC tags used to work do not anymore (that window inviting the user is not popping) => A reboot is necessary to bring back the readings within those « perfectly » working nfc apps. 
From these remarks, my conclusion is that the problem is related to the nfc manager package in combinaison with the new IOS.

kensang, 2021-01-28:
I seem to have a similar issue, my finding after some testing is as followed:

1) My testing device is iPhone XS Max 14.3

2) Using the native iOS demo janlionly/NFCReaderWriter works on my testing device

3) Using exactly the same plist and entitlement as the working native iOS demo (in point #2), and using this flutter NFC manager package doesn't make the « hold the phone near the tag» window show up

Since the native iOS demo janlionly/NFCReaderWriter works, I am guessing there are ways that makes the flutter package works.  But I am also guessing that's there are issues/bugs for iOS 13/14 that the developer has to work around.  It's supposed to be easy but it looks like it's sort of buggy or difficult to get the tag UID as you can find a lot of questions about that on the internet with no real solutions.

okadan, 2021-01-28:
I tested the behavior with a minimal app created with "Flutter+nfc_manager" and "Pure Swift" respectively.

As a result, I have confirmed that the exactly same behavior on both.

The following is a fragment of the Swift code and  the detailed behavior of the test.

**Environment**

iPhone7 / iOS14.3 real device.
Xcode 12.4
Flutter 1.25.0-8.3.pre (beta channel)

**Swift Code**

```swift
import CoreNFC
import SwiftUI

struct ContentView: View {
    var body: some View {
        Button(action: {
            let manager = NFCManager()
            let session = NFCTagReaderSession(pollingOption: NFCTagReaderSession.PollingOption.iso14443, delegate: manager)
            session!.begin()
        }) {
            Text("Start Session")
        }
    }
}

class NFCManager: NSObject, NFCTagReaderSessionDelegate {
    func tagReaderSessionDidBecomeActive(_ session: NFCTagReaderSession) {}
    
    func tagReaderSession(_ session: NFCTagReaderSession, didInvalidateWithError error: Error) {}
    
    func tagReaderSession(_ session: NFCTagReaderSession, didDetect tags: [NFCTag]) {}
}

```

**Detailed Behavior**

1. Press the "Start Session" button
    => [OK] The popup appers.
2. Cancel Popup
    => [OK] The popup disappers.
3. Press the "Start Session" button again
    => [NG] No response. The log is empty.
4. Press the "Start Session" button again
    => [NG] Popup appears but get the error log.
    ```
    [CoreNFC] 00000002 80a45dd0 -[NFCTagReaderSession beginSessionWithConfig:]:365  error:Error Domain=NFCError Code=203 "System resource unavailable" UserInfo={NSLocalizedDescription=System resource unavailable}, errorCode: 0xcb
    ```
5. 1 to 4 endlessly.

ayyoubelamrani4, 2021-01-28:
> I tested the behavior with a minimal app created with "Flutter+nfc_manager" and "Pure Swift" respectively.
>
> As a result, I have confirmed that the exactly same behavior on both.
>
> The following is a [...]

I guess you have to invalidate your session after finishing your reading, with something like (session.end() right after).
=>The issue is not with the Swift code but with the **flutter nfc manager** one combined with the new IOs 14.3, there is some weird behaviour there.

kensang, 2021-02-01:
I happened to find a flutter package that works for me: flutter_nfc_kit , see if you can use it as a reference to fix this issue.

ayyoubelamrani4, 2021-02-02:
```
channel.invokeMethod('Nfc#startSession', {
      'pollingOptions':
          pollingOptions.map((e) => $NfcPollingOptionTable[e]).toList(),
      'alertMessage': alertMessage,
    });
```
Calling this method from the debugger sig kills the app => which probably starts a session and returns an error hence the session would never be Interrupted and the readings are never done. 
I am not sure about what is happening when the channel invokes the method, @okadan any hints? would this help?

okadan, 2021-02-04:
After some investigation, I got a few hints.

It seems that this issue occurs when `NFCTagReaderSession.PollingOption.iso18092` is specified but `com.apple.developer.nfc.readersession.felica.systemcodes` is not.

This is a specification of the iOS system.

Both Flutter users and Swift users have the same issue.

**Workaround**

1. If you don't need to detect the iso18092 tags, remove the `iso18092` option from `pollingOptions`.

```dart
NfcManager.instance.startSession(
  // Here.  NOTE: all options are specified by default.
  pollingOptions: {NfcPollingOption.iso14443, NfcPollingOption.iso15693},
  onDiscovered: ...
);
```

2. If you need to detect the iso18092 tags, specify the `iso18092` option, but also specify the `com.apple.developer.nfc.readersession.felica.systemcodes` to your `Info.plist`. ([example](https://github.com/okadan/flutter-nfc-manager-app/blob/master/ios/Runner/Info.plist#L48))

closed 2021-02-18 by okadan

yetunnaing18121996, 2022-02-17:
The IOS NFC Scan Dialog doesn't appear and I got the error message like this, 
[CoreNFC] 00000002 800c0c60 -[NFCTagReaderSession beginSessionWithConfig:]:365  error:Error Domain=NFCError Code=202 "Session invalidated unexpectedly" UserInfo={NSLocalizedDescription=Session invalidated unexpectedly}, errorCode: 0xca.
I've already setup all the necessary requirements. 

I used:
Flutter sdk: Flutter 2.8.1
nfc_manager: ^3.1.0
device: iPhone 7
Os Version: 14.4.2