okadan/flutter-nfc-manager #40: Is it possible to listen on NFC availability?

MassimoTambu, 2021-03-01:
Hi, thanks for your work!

I would ask you if it could be possible to listen on NFC status changes (ON - OFF). I didn't find any reference to this in the code but I could be wrong.. Is it something that you already implemented? If not, it could be possible to do so?

I would like to show different screens based on the current NFC status and I don't think it's a good idea calling every 'n' seconds isAvailable() method.

Thanks in advance!

Max

barangungor, 2021-12-24:
I want do same thing. Could you?

okadan, 2021-12-24:
Currently not supported.

But we could implement a dart stream receives the [ACTION_ADAPTER_STATE_CHANGED](https://developer.android.com/reference/android/nfc/NfcAdapter#ACTION_ADAPTER_STATE_CHANGED). (Android only)

HeropolisDa2ny, 2024-04-03:
Is there a roadmap planned for this feature ?

okadan, 2025-09-28:
This has been implemented in v4.0.0 or later.

[NfcManagerAndroid.instance.onStateChanged](https://pub.dev/documentation/nfc_manager/latest/nfc_manager_android/NfcManagerAndroid/onStateChanged.html)

closed 2025-09-28 by okadan