okadan/flutter-nfc-manager #43: Issue with iOS 14.4.1 on iPhone 7

torgebauer, 2021-03-19:
I tried out your demo code and also checked out your demo application on my iPhone 7 iOS 14.4.1 and it is not working.

`2021-03-19 11:28:29.884901+0100 Runner[403:8791] [CoreNFC] 00000002 83700c20 -[NFCHardwareManager queueReaderSession:sessionConfig:completionHandler:]:96  error=Error Domain=NFCError Code=203 "System resource unavailable" UserInfo={NSLocalizedDescription=System resource unavailable}
2021-03-19 11:28:29.885297+0100 Runner[403:8791] [CoreNFC] 00000002 80558870 -[NFCTagReaderSession beginSessionWithConfig:]:365  error:Error Domain=NFCError Code=203 "System resource unavailable" UserInfo={NSLocalizedDescription=System resource unavailable}, errorCode: 0xcb`

This is the exception with your example code.

JunOkumura, 2021-03-25:
I have also experienced same issue.
I used flutter version 1.22.x and nfc_manager version 2.0.3 when it is happened.

I tried use flutter version 2.0.3 and nfc_manager version 3.0.0+1.
my sample works fine.

But demo code think need fix for nfc_manager 3.0.0 or later usage.

MochiMochiTosiki, 2021-03-25:
I rebooted of mac and that fixed it.

JunOkumura, 2021-03-25:
I have experienced this error using other sample.

I reboot PC and iPhone.
it works again.

okadan, 2021-03-25:
This may be an iOS-specific issue.

In my experience, the exact same thing happened with a pure Swift project.

We will also need to check #38.

closed 2021-03-29 by okadan