okadan/flutter-nfc-manager #45: iOS iso7816 sendCommand Response Error

whywilson, 2021-04-29:
I'm working on reading an iso1443-4 card on iOS. It works well on Swift code but failed in the flutter project.

## Code with flutter-nfc-manager
```
iso7816Tag.sendCommandRaw(Uint8List.fromList([0x00, 0xA4, 0x04, 0x00, 0x08, 0x50, 0x41, 0x59, 0x2E, 0x54, 0x49, 0x43, 0x4C, 0x00])).then((value) =>
{
   print("select App response ${value}")
});
```
## flutter-nfc-manager Response
```
flutter: select App response []
[CoreNFC] 00000002 8001cb40 -[NFCTagReaderSession setAlertMessage:]:90  (null)
```

The command is correct, it works well in Xcode with swift.
## Code with Swift
```
let selectApp : NFCISO7816APDU = NFCISO7816APDU(data: Data.init([0x00, 0xA4, 0x04, 0x00, 0x08, 0x50, 0x41, 0x59, 0x2E, 0x54, 0x49, 0x43, 0x4C, 0x00]))!
            
session.connect(to: tag0!) { (e: Error?) in tag?.sendCommand(apdu: selectApp, completionHandler: { (data: Data, sw1: UInt8, sw2: UInt8, error: Error?) in
  print("selectApp APDU Result \(data.hexDescription)")
})
}
```
## Swift Response
```
selectApp APDU Result 6f3484085041592e5449434ca5289f0801029f0c21ffffffffffffffff000000000000000000000000000000002019082400000186a0
```

Please help to confirm the issues. 🙏

okadan, 2021-04-29:
@whywilson Please tell me the version of the plugin.

I recommend to use the newer version as much as possible.

whywilson, 2021-04-29:
> @whywilson Please tell me the version of the plugin.
> 
> I recommend to use the newer version as much as possible.

It's always the latest version.
I guess the tag used to connect was not converted to ISO7816TAG.
Here's the full swift code which works well.

```
        func tagReaderSession(_ session: NFCTagReaderSession, didDetect tags: [NFCTag]) {
            print("didDetect \(tags)")
            var nfcIso7816Tag: NFCISO7816Tag? = nil
            var nfcTag: NFCTag? = nil
            
            for tag in tags {
                if case let .iso7816(cTag) = tag {
                    nfcIso7816Tag = cTag
                    nfcTag = tag
                }
            }
            
            if nfcIso7816Tag == nil {
                session.invalidate(errorMessage: "Card not support")
            }
            
            let selectApp : NFCISO7816APDU = NFCISO7816APDU(data: Data.init([0x00, 0xA4, 0x04, 0x00, 0x08, 0x50, 0x41, 0x59, 0x2E, 0x54, 0x49, 0x43, 0x4C, 0x00]))!
            let readPurse : NFCISO7816APDU = NFCISO7816APDU(data: Data.init([0x80, 0x5C, 0x00, 0x02, 0x04]))!
                                   ⬇️
            session.connect(to: nfcTag!) { (e: Error?) in
                         ⬇️
                nfcIso7816Tag?.sendCommand(apdu: selectApp, completionHandler: { (data: Data, sw1: UInt8, sw2: UInt8, error: Error?) in
                    print("selectApp APDU Result \(data.hexDescription)")
                })
                
                nfcIso7816Tag?.sendCommand(apdu: readPurse, completionHandler: { (data: Data, sw1: UInt8, sw2: UInt8, error: Error?) in
                    let moneyString  = data.hexDescription
                    let moneyValue = Double(Int(moneyString, radix: 16)!) / 100.0
                    print("Data \(data.hexDescription)")
                    session.alertMessage = "羊城通 余额：￥\(moneyValue)"
                })
            }
        }
```

okadan, 2021-04-29:
> flutter: select App response []

@whywilson 

It seems that the `sendCommandRaw` returns a list, so I guessed the version is old. (`< 2.0.0`)

But you say that you're using the latest version, so this is probably the plugin bug.

I will investigate in the near the future.

whywilson, 2021-04-29:
> > flutter: select App response []
> 
> @whywilson
> 
> It seems that the `sendCommandRaw` returns a list, so I guessed the version is old. (`< 2.0.0`)
> 
> But you say that you're using the latest version (`>= 3.0.0`), so this is probably the plugin bug.
> 
> I will investigate in the near the future.

Thank you for the work. Please also check here. https://github.com/nfcim/flutter_nfc_kit/issues/32

whywilson, 2021-05-08:
This issue resulted from the old batch of the transport card that I test. It works after the Master File is selected manually. 
So this issue can be closed now.
Thank you for the work.

closed 2021-05-08 by whywilson