okadan/flutter-nfc-manager #47: Feature Request: Password Protected Tags & Read-Only Tags

JaffaKetchup, 2021-05-15:
Hello all,
I think a rather useful feature would be to be able to password protect new/existing tags or create read only tags. This is supported by some NDEF formatted cards and would allow these tags to be placed in public places, without making them vulnerable to sabotage or even accidental writing.
Password protected tags would require the password to be removed to make them writable (they'd always be readable). Read only tags would never be able to be written again.
Thanks, and I hope this idea does get implemented because it is a main idea of my app.

JaffaKetchup, 2021-05-15:
Ok, so I've just downloaded the example app, and it looks like there is a permeant locking feature that works. Great! Not so great when there is no warning that that's what it's going to do and there is no way to undo it. **Please add a warning (fix the ListFooter widget), because now I have a locked tag with no data on it.** 

But anyway, that's useful. Is it possible to have password protected tags, because that is supported by some tags, and would be more useful, and undoable!?

okadan, 2021-05-17:
Duplicate of #20

closed 2021-05-17 by okadan