okadan/flutter-nfc-manager #51: i got error java.lang.RuntimeException: Tag incorrectly enumerated as MIFARE Classic, SAK = 32

sabardotnet, 2021-06-12:
D/AndroidRuntime(22979): Shutting down VM
E/AndroidRuntime(22979): FATAL EXCEPTION: main
E/AndroidRuntime(22979): Process: com.jm.travoy, PID: 22979
E/AndroidRuntime(22979): java.lang.RuntimeException: Tag incorrectly enumerated as MIFARE Classic, SAK = 32
E/AndroidRuntime(22979): 	at android.nfc.tech.MifareClassic.<init>(MifareClassic.java:200)
E/AndroidRuntime(22979): 	at android.nfc.tech.MifareClassic.get(MifareClassic.java:138)
E/AndroidRuntime(22979): 	at io.flutter.plugins.nfcmanager.TranslatorKt.getTagMap(Translator.kt:88)
E/AndroidRuntime(22979): 	at io.flutter.plugins.nfcmanager.NfcManagerPlugin$handleNfcStartSession$1$1.run(NfcManagerPlugin.kt:110)
E/AndroidRuntime(22979): 	at android.os.Handler.handleCallback(Handler.java:883)
E/AndroidRuntime(22979): 	at android.os.Handler.dispatchMessage(Handler.java:100)
E/AndroidRuntime(22979): 	at android.os.Looper.loop(Looper.java:214)
E/AndroidRuntime(22979): 	at android.app.ActivityThread.main(ActivityThread.java:7540)
E/AndroidRuntime(22979): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(22979): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
E/AndroidRuntime(22979): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:980)
I/Process (22979): Sending signal. PID: 22979 SIG: 9
Lost connection to device.

sabardotnet, 2021-06-15:
any one can answer my question ?

okadan, 2021-06-15:
@sabardotnet 

Android seems to use SAK to distinguish between different tags, but it doesn't catch all SAKs. 
(Exception is thrown in [MifareClassic.java](https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/nfc/tech/MifareClassic.java))

This plugin may need to skip the MifareClassic emulation in this case.

sabardotnet, 2021-06-23:
i don't know why after upgrade flutter SDK suddenly get error like this, before upgrade no error. what about it depend on NFC hardware of mobile ?

sabardotnet, 2021-06-23:
i use other device as tester, more cheaper than other device its missing.

sarmiladhandapani, 2021-07-12:
same issue here

sabardotnet, 2021-07-26:
i've tested with more expensive device, the plugin working as well. how it can happen ??

dididudikov, 2022-01-27:
The best solution to this problem is to provide the option to turn off Mifare Classic emulation. If the card is detected to have Iodep and Mifare classic technology, then never activate the Mifare Classic clas.

Here is the source that has been modified so that there is an option to turn off mifare emulation

[nfc_manager-3.1.0.zip](https://github.com/okadan/flutter-nfc-manager/files/7949954/nfc_manager-3.1.0.zip)
[nfcmanager.zip](https://github.com/okadan/flutter-nfc-manager/files/7949955/nfcmanager.zip)
s

dorklord23, 2024-06-12:
has this issue been fixed?

dorklord23, 2024-06-12:
> The best solution to this problem is to provide the option to turn off Mifare Classic emulation. If the card is detected to have Iodep and Mifare classic technology, then never activate the Mifare [...]

How are others supposed to used the patched files?