okadan/flutter-nfc-manager #52: [ios] Unhandled Exception: PlatformException(unavailable, Only available in iOS 13.0 or newer, null, null)

aoatmon, 2021-06-16:
### given 
- flutter create nfc_ios
- add to ios\runner the file `Runner.entitlements` with 

```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>com.apple.developer.nfc.readersession.formats</key>
	<array>
		<string>TAG</string>
	</array>
</dict>
</plist>
```
- add to plist 

```xml

	<!-- added for nfc START -->
	<key>NFCReaderUsageDescription</key>
	<string>Read NFC Tags</string>
	<!-- added for nfc STOP -->
	
```

- add code

```dart
import 'package:flutter/material.dart';
import 'package:nfc_manager/nfc_manager.dart';

void _justPrint() async {
  final instance = NfcManager.instance;
  if (await instance.isAvailable()) {
    instance.startSession(onDiscovered: (tag) async {
      for (final entry in tag.data.entries) {
        print('key ${entry.key}');
        print('value ${entry.value}');
      }
      print(tag.handle);
    });
  }
}

void main() => runApp(const MaterialApp(home: JustListen()));

class JustListen extends StatefulWidget {
  const JustListen({Key? key}) : super(key: key);

  @override
  _JustListenState createState() => _JustListenState();
}

class _JustListenState extends State<JustListen> {
  @override
  void initState() {
    _justPrint();
    super.initState();
  }

  @override
  Widget build(BuildContext context) {
    return const Material(child: Center(child: CircularProgressIndicator()));
  }
}

```

[... 153 lines cut ...]
name: nfc_ios
description: A new Flutter project.
version: 1.0.0+1

environment:
  sdk: ">=2.12.0 <3.0.0"

dependencies:
  flutter:
    sdk: flutter
  nfc_manager: ^3.1.0

dev_dependencies:
  flutter_test:
    sdk: flutter

flutter:
  uses-material-design: true
```

</details>

### note

same code works perfectly on android

okadan, 2021-06-16:
> Only available in iOS 13.0 or newer

Check your iOS version.

aoatmon, 2021-06-16:
> > Only available in iOS 13.0 or newer
> 
> Check your iOS version.
@okadan 
so is meant not to support older versions?

closed 2021-06-16 by aoatmon