okadan/flutter-nfc-manager #55: Error:nfc.platformexception(io_exception,null,null,null)

juniorliga, 2021-07-23:
Hi, I have this problem with this packages in android. On the various devices it sometimes works but other times it fails. the message that comes out to me is this: nfc.platformexception(io_exception,null,null,null).

Thank you.

SmartNodeAutomation, 2021-08-09:
Hi, I also found this error when I write data that is exceeding its length then it will show this error. I erase data and format my tag then after it will work.

Durgesh9696, 2021-08-09:
Hi, I also get this error when I write data that is exceeding its length then it will show this error. I erase data and format my tag then after it will work.

DiWizard, 2021-08-19:
Hello!
It looks like it happens (at least in my case) when I try to write a new empty unused label for the first time. After clear label using "NFC Tools" always works fine...

DasElias, 2021-09-09:
I get this error even after formatting the tag. When I try to write that particular data not with my Android phone, but rather with an app on my iPhone, I don’t get that exception.

mnurtay, 2021-09-13:
Hello!
I used the Mifare card. When I try call "authenticateSectorWithKeyA" method, I get this error: PlatformException(invalid_parameter, Tag is not found, null, null)

syahrizalakbar, 2022-12-29:
same here, happened to me too

batoul-alani, 2023-02-02:
is there any solution for this issue ?

mfundolukhwazi, 2023-07-17:
Hello, 
did you guys find a solution to PlatformException(invalid_parameter, Tag is not found, null, null)?

mfundolukhwazi, 2023-07-18:
After testing out this [demo app](https://play.google.com/store/apps/details?id=com.naokiokada.nfcmanager) made from this library, on the Mifare classic card details there is a timeout of 618 which I didn't know about. I was making an asynchronous call after discovering the tag to check if doesn't already exist in the db and this ate up the time and by the time i was formatting the tag to ndef, it had timed out hence the exception.

etrnljackson, 2024-01-16:
I was also getting the "Tag is not found" error. In my case, I had a method inside `onDiscovered` that I wasn't `await`ing. After adding `await`, I stopped getting the error.

sushant-shidore, 2024-07-02:
@etrnljackson Thanks a lot for your comment. I was facing this issue since long, and then checked all methods being called inside onDiscovered. One of them was missing `await`. Added it and the issue just went off.

hebaothman, 2025-01-05:
Same here any solution. The issue occurs when i try to write on NFCA tag