okadan/flutter-nfc-manager #58: Question: ISO-14443A (NTAG213) support for both android & ios

makhosravi, 2021-07-30:
Hi

I'm going to start a flutter cross-platform project and use your package as NFC manager.
As for the NFC tag, I'm going to use the NTAG213 chipset (ISO-14443-A), just want to read/write an URL on it, and I need to know that your package can support this tag and chipset on both android and ios (Right now NFC tag is not available for me to test your package).

It would be so nice to get your response soon.
With regards

okadan, 2021-07-30:
> I'm going to use the NTAG213 chipset (ISO-14443-A), just want to read/write an URL on it, and I need to know that your package can support this tag and chipset on both android and ios

Supported.

The `Ndef` class will do it.

makhosravi, 2021-07-30:
Thank you for your quick response, Have a great day

closed 2021-07-30 by okadan

makhosravi, 2021-08-03:
Hi
I have another question.
Is it possible to make the tag password protected (if possible on a tag like NTAG213), so the tag becomes personal to a specific user, with this package?