okadan/flutter-nfc-manager #68: IOS not working

suiebay, 2021-11-09:
Hi,
I created a project and for android is working well, but with iOS i start reading nfc tags, I see the system window to scan, but nothing happens, its not working and after some seconds I get a time out and the system closes the window.
Could you help me please?
thanks so much

okadan, 2021-11-09:
Poor information, but It seems that your target tag type is not listed in Info.plist file.

Check the following properties:

* [com.apple.developer.nfc.readersession.felica.systemcodes](https://developer.apple.com/documentation/bundleresources/information_property_list/systemcodes)
* [com.apple.developer.nfc.readersession.iso7816.select-identifiers](https://developer.apple.com/documentation/bundleresources/information_property_list/select-identifiers)

Also check the example [Info.plist](https://github.com/okadan/nfc-manager/blob/master/ios/Runner/Info.plist) in my app.

suiebay, 2021-11-09:
Thank you for your response, but I listed all the types and did IOS Setup as here [nfc_manager](https://pub.dev/packages/nfc_manager)

Here you can see images of my info.plist file:
[image]

[image]

Here you can see images of my XCode Capability:
[image]

Here you can see images of my Entitlement file:
[image]

When I run app on IOS Device, dialogue screen of NFC is opens but it can't read the NFC tag, but with android same tag works. Could you please help me, where problem could be?

okadan, 2021-11-09:
What is your target tag type?

If your tag type is Iso7816, you need to list the AID of that tag to read it.

suiebay, 2021-11-10:
Could you please help me, how can I list the AID and where i can get it?

okadan, 2021-11-10:
Find an AID of ISO7816 tag yourself, then list it in `com.apple.developer.nfc.readersession.iso7816.select-identifiers` section in your `Info.plist`

suiebay, 2021-11-10:
I added all the AIDs but still can't read, also cloned [example project](https://github.com/okadan/flutter-nfc-manager) and run, it can read tag in Android device but not in ios :( What could be the problem?

redDwarf03, 2021-12-25:
@suiebay could you share your project ?

rxIndrajeet, 2022-03-03:
@suiebay did you find the solution for this issue? I am also facing the same issue

Hesam-Taleghani, 2022-03-27:
Thank you for this great project, do you have a plan to add iOs support for NfcA tags? @okadan

magicmatt007, 2022-04-23:
@okadan:

I only have a free developer account and after some fiddling around and reading below documentation from Apple, I came to the conclusion that a PAID Apple Developer Account (ADP) is required to use NFC on IOS. Do you agree? If yes, maybe add that hint to the read me of this great plugin :-)

https://help.apple.com/developer-account/#/dev21218dfd6

Matthias

closed 2025-09-28 by okadan