okadan/flutter-nfc-manager #71: After NFC tag is scanned on Android device, popup "New tag collected" appears over the Flutter application. How to prevent it?

mateoKutnjak, 2021-11-28:


mateoKutnjak, 2021-11-28:
I found the solution.

Android activity should be modified. Look at the [https://github.com/okadan/nfc-manager/blob/master/android/app/src/main/kotlin/com/naokiokada/nfcmanager/MainActivity.kt](url) file.

closed 2021-11-28 by mateoKutnjak

ntatko, 2022-09-10:
Just to make the solution a little more visible - you need to override the default android behavior, and change some settings when the app pauses and wakes.

In `MainActivity.kt`, you'll need to import 

```kt
import android.app.PendingIntent
import android.content.Intent
import android.nfc.NfcAdapter
```

and inside of your main activity, add the following code:

```kt
    override fun onResume() {
        super.onResume()
        val intent = Intent(context, javaClass).addFlags(Intent.FLAG_ACTIVITY_SINGLE_TOP)
        val pendingIntent = PendingIntent.getActivity(context, 0, intent, 0)
        NfcAdapter.getDefaultAdapter(context)?.enableForegroundDispatch(this, pendingIntent, null, null)
    }

    override fun onPause() {
        super.onPause()
        NfcAdapter.getDefaultAdapter(context)?.disableForegroundDispatch(this)
    }
```

claeszacho, 2022-10-12:
Could this be added to the documentation. I spent a lot of time and frustrations on the issue, until I found this solution. Thanks, anyway.

adplant, 2023-06-05:
Does anyone know about how to prevent this on iOS as well? The solution above is working for Android, but iOS still populates the message showing that you can open the link (represented in the ndef record).