okadan/flutter-nfc-manager #76: read nfc card more than one time in one session!!

taha-elkholy, 2021-12-27:
when using NfcManager.instance.startsession it read the card many times 
I am using NfcManager.instance.stopsessionafter reading the card to finish reading
 await NfcManager.instance.startSession(
          onDiscovered: (NfcTag tag) async {
            // stop Session
            NfcManager.instance.stopSession();
            if (kDebugMode) {
              print('Tag data record: ${tag.data}');
            }
            emit(ReadNFCSuccessState(subWallet));
          }

v1lling, 2021-12-28:
I think that's the expected behavior.

taha-elkholy, 2021-12-28:
but when I use the onDiscover to navigate to another screen it also navigates to the same screen many times 
how to take only the first successful reading and deal with it one time? no many times

v1lling, 2021-12-28:
Just like you did, using NfcManager.instance.stopSession().
You can have a look at the [example](https://github.com/okadan/flutter-nfc-manager/blob/master/example/lib/main.dart) where the author of this plugin did the same thing.