okadan/flutter-nfc-manager #77: how to convert the tag.data to usable data

taha-elkholy, 2022-01-07:
I have an NFC card which writes on it a URL,
But when trying to read it, It give me a tag.data not readable for me like this

{id: [249, 109, 254, 163], nfca: {atqa: [4, 0], maxTransceiveLength: 253, sak: 8, timeout: 618}, mifareclassic: {blockCount: 64, maxTransceiveLength: 253, sectorCount: 16, size: 1024, timeout: 618, type: 0}, ndef: {cachedMessage: {records: [{typeNameFormat: 1, type: [84], identifier: [], payload: [2, 101, 110, 72, 101, 108, 108, 111, 32, 87, 111, 114, 108, 100, 33]}, {typeNameFormat: 1, type: [85], identifier: [], payload: [4, 102, 108, 117, 116, 116, 101, 114, 46, 100, 101, 118]}, {typeNameFormat: 2, type: [116, 101, 120, 116, 47, 112, 108, 97, 105, 110], identifier: [], payload: [72, 101, 108, 108, 111]}, {typeNameFormat: 4, type: [99, 111, 109, 46, 101, 120, 97, 109, 112, 108, 101, 58, 109, 121, 116, 121, 112, 101], identifier: [], payload: [109, 121, 100, 97, 116, 97]}]}, canMakeReadOnly: false, isWritable: true, maxSize: 716, type: com.nxp.ndef.mifareclassic}}

how to get the message and how to use it correctly?

okadan, 2022-01-24:
@taha-elkholy 

See [API Doc](https://pub.dev/documentation/nfc_manager/latest/nfc_manager/NfcTag/data.html).

For example:

```dart
final ndef = Ndef.from(tag);

if (ndef == null)
  print('Tag is not ndef.');
  return;

print(ndef.cachedMessage);
```

taha-elkholy, 2022-01-24:
Thanks for the replay,
for me, I've used this
it works well

  

    Ndef? ndef = Ndef.from(tag);
    if (ndef != null) {
      NdefMessage? message = ndef.cachedMessage;
      if (message != null) {
        List<NdefRecord> records = message.records;
        NdefRecord record = records[0];

        Uint8List payload = record.payload;
        // can not print it put can show it on screen
        String payloadAsString = String.fromCharCodes(payload);
        return payloadAsString ;
      } else {
        // message null
        emit(ReadNFCErrorState('Ndef Message is null'));
        if (kDebugMode) {
          print('message null $message');
        }
        return null;
      }
    } else {
      // ndef null
      emit(ReadNFCErrorState('Ndef is null'));
      if (kDebugMode) {
        print('ndef null $ndef');
      }
      return null;
    }

closed 2022-02-07 by okadan