okadan/flutter-nfc-manager #88: PlatformException(io_exception, null, null, null) when trying to perform readPages

Matix-Media, 2022-03-13:
Hello, I'm currently trying to read raw data from an MifareUltralight card. But when I try to perform `tag.readPages(pageOffset: 0x04)` (for the 5th page, as the first 4 are reserved for manufacturer information as far as I know) I get the following error:
```
E/flutter (26164): [ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: PlatformException(io_exception, null, null, null)
E/flutter (26164): #0      StandardMethodCodec.decodeEnvelope
package:flutter/…/services/message_codecs.dart:607
E/flutter (26164): #1      MethodChannel._invokeMethod
package:flutter/…/services/platform_channel.dart:177
E/flutter (26164): <asynchronous suspension>
E/flutter (26164): #2      _SavedCardsList._addMifareUltralightCard
package:nfc_card_emulator/widgets/saved_cards_list.dart:96
E/flutter (26164): <asynchronous suspension>
E/flutter (26164): #3      _SavedCardsList._addNfcCard
package:nfc_card_emulator/widgets/saved_cards_list.dart:85
E/flutter (26164): <asynchronous suspension>
E/flutter (26164): #4      _SavedCardsList._detectNfcCard.<anonymous closure>
package:nfc_card_emulator/widgets/saved_cards_list.dart:75
E/flutter (26164): <asynchronous suspension>
E/flutter (26164): #5      NfcManager._handleOnDiscovered
package:nfc_manager/…/nfc_manager/nfc_manager.dart:104
E/flutter (26164): <asynchronous suspension>
```

I don't really know what the problem is as I can get other information provided by the MifareUltralight class and I know that my device suports it as I can see when printing out the raw data information from NfcTag.data which shows a mifare ultralight section.

Can someone help me with my error as I don't know any further.

simonloewe, 2022-03-24:
Same issue here

simonloewe, 2022-03-25:
Solved. This is not an issue with the lib on my side but had to do with wrong implementation.

CharlesAMiller, 2022-04-07:
@simonloewe Do you recall what you changed, or what your correct implementation involved? I'm running into similar problems as OP, and I imagine there's likely some user error on my part.

simonloewe, 2022-04-07:
I had this issue reading MifareUltralight tags. Same error got thrown. Turned out that 1) the hex code I was sending was wrong. This one here `Uint8List.fromList([0x30, 52])` reads out the chip at offset 52. Then secondly I used the wrong method. I used this here: `await tech.sendMiFareCommand(Uint8List.fromList([0x30, 52]))` . I'm quite certain that this error gets thrown when the hexcode is wrong or you use the wrong function. As you can see [here](https://github.com/okadan/flutter-nfc-manager/blob/master/lib/src/platform_tags/mifare.dart#L43), there are 3 methods for Mifare at your disposal. Only one worked for me.