okadan/flutter-nfc-manager #91: iOS: Missing required entitlement

aakashr143, 2022-04-11:
The code works fine with android, but on iOS it is giving 'Missing required entitlement'

my entitlement file, it matches with the example provided
`
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>com.apple.developer.nfc.readersession.formats</key>
	<array>
		<string>NDEF</string>
		<string>TAG</string>
	</array>
</dict>
</plist>
`

Info.plist
`
<key> NFCReaderUsageDescription</key>
	<string>>This app uses the NFC to read the tags</string>
	<key>NFCReaderUsageDescription</key>
	<string>NFC Usage Description</string>
	<key> com.apple.developer.nfc.readersession.felica.systemcodes</key>
	<array>
		<string>0003</string>
		<string>04D1</string>
		<string>8008</string>
		<string>80DE</string>
		<string>865E</string>
		<string>8592</string>
		<string>8B5D</string>
		<string>8FC1</string>
		<string>FE00</string>
	</array>
	<key>com.apple.developer.nfc.readersession.iso7816.select-identifiers </key>
	<array>
		<string>A0000002310100000000000000000000</string>
		<string>A0000002310200000000000000000000</string>
		<string>A0000002480300000000000000000000</string>
		<string>A00000006510</string>
		<string>A0000000651010</string>
		<string>315041592E5359532E4444463031</string>
		<string>325041592E5359532E4444463031</string>
		<string>44464D46412E44466172653234313031</string>
		<string>A00000000101</string>
		<string>A000000003000000</string>
		<string>A00000000300037561</string>
		<string>A00000000305076010</string>
		<string>A0000000031010</string>
		<string>A000000003101001</string>
		<string>A000000003101002</string>
		<string>A0000000032010</string>
		<string>A0000000032020</string>
		<string>A0000000033010</string>
		<string>A0000000034010</string>
		<string>A0000000035010</string>
		<string>A000000003534441</string>
		<string>A0000000035350</string>
		<string>A000000003535041</string>
		<string>A0000000036010</string>
		<string>A0000000036020</string>
		<string>A0000000038002</string>
		<string>A0000000038010</string>
		<string>A0000000039010</string>
		<string>A000000003999910</string>
		<string>A0000000040000</string>
		<string>A00000000401</string>
		<string>A0000000041010</string>
		<string>A00000000410101213</string>
[... 237 lines cut ...]
		<string>E828BD080F</string>
		<string>F0000000030001</string>
	</array>
	<key>NSLocalNetworkUsageDescription</key>
	<string>Local network usage descrption</string>
	<key>NSBonjourServices</key>
	<array>
		<string>_dartobservatory._tcp</string>
	</array>
`

function for reading nfc
`
void startNFCSession() async {
    NfcManager.instance.startSession(
        alertMessage: "iOS Alert Message",
        onDiscovered: (tag) async {
          Ndef? ndef = Ndef.from(tag);
          NfcManager.instance.stopSession();
        },
        onError: (err) async {
          print(err.message);
        });
  }
`

aakashr143, 2022-04-11:
@okadan still the same error

Dooks123, 2022-04-20:
`<string>>This app uses the NFC to read the tags</string>` this part in the first line of your `info.plist` you have `>>` maybe try and remove that?

aakashr143, 2022-04-21:
> `<string>>This app uses the NFC to read the tags</string>` this part in the first line of your `info.plist` you have `>>` maybe try and remove that?

That's not the start of the `info.plist`. That the point from where the nfc part is starting

Dooks123, 2022-04-21:
@aakashr143 Well, I can't get iOS to scan any NFC tags at all as well.

HenrikH96, 2022-11-18:
anything new on that topic ? 
i got the same Missing required entitlements"error

Dooks123, 2022-11-18:
@HenrikH96 Here you go:

In **Info.plist**:

    <?xml version="1.0" encoding="UTF-8"?>
    <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
    <plist version="1.0">
    <dict>
        
        <key>NFCReaderUsageDescription</key>
        <string>[YourApp] needs access to read and write NFC [BeSpecific] tags</string>
        <key>com.apple.developer.nfc.readersession.felica.systemcodes</key>
        <array>
            <string>12FC</string>
            <string>0000</string>
        </array>
        <key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
        <array>
            <string>D2760000850100</string>
            <string>D2760000850101</string>
            <string>D2760001180101</string>
            <string>00000000000000</string>
        </array>
    </dict>
    </plist>

In **Runner.entitlements** (You need to add entitlements file with XCode):

    <?xml version="1.0" encoding="UTF-8"?>
    <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
    <plist version="1.0">
    <dict>
        <key>com.apple.developer.nfc.readersession.formats</key>
        <array>
            <string>TAG</string>
        </array>
    </dict>
    </plist>

Don't ask me why it works, because I can't decipher Apple's documentation either.

HenrikH96, 2022-11-18:
@Dooks123 
Thank you very much, it seems like 
com.apple.developer.nfc.readersession.felica.systemcodes and/or com.apple.developer.nfc.readersession.iso7816.select-identifiers are needed and not optional. 
This was the only thing I changed to get it working. 
The Error "missing required entitlements" is a bit misleading because in fact missing keywords in info.plist caused the error.

Dooks123, 2022-11-18:
@HenrikH96 Glad it helped.

closed 2024-01-23 by okadan