okadan/flutter-nfc-manager #93: This just only works using a TAG?

byronfrancisco, 2022-04-28:
I want to create an application working between two smarthphones, and Im not sure if I can use this widget or if is better another.

MariosKoni, 2022-05-09:
I want to use this package for the same reason as well. Will it work?

sammusaev, 2022-08-04:
As far as I am aware, Android [does not allow ](https://developer.android.com/guide/topics/connectivity/nfc/nfc)ndef data transfers between Android devices (i.e. having a device act as a tag and reader). I think you'd have to use Android Beam or NeabySharing for that.

byronfrancisco, 2022-08-04:
Ive found a solution. But One of them needs be a Host Card Emulation, https://www.zeromolecule.com/blog/host-card-emulation-hce-with-android-and-flutter/ , this is the link if you want to do in Flutter.
And the reader can use the pluglin nfc_manager.