okadan/flutter-nfc-manager #101: Keep listening for cards after the screen is off and on

moham96, 2022-08-08:
Hi,
when the app is opened and a session is started and listening for cards if I lock the screen off and on again then the app is no longer listening for cards, to reproduce use this simple app:

```
import 'package:flutter/material.dart';
import 'package:nfc_manager/nfc_manager.dart';
import 'package:nfc_manager/platform_tags.dart';

void main() {
  runApp(const MaterialApp(
    home: HomeWidget(),
  ));
}

class HomeWidget extends StatefulWidget {
  const HomeWidget({Key? key}) : super(key: key);

  @override
  State<HomeWidget> createState() => _HomeWidgetState();
}

class _HomeWidgetState extends State<HomeWidget> {
  List<int> cards = [];

  void nfcInit() async {
    try {
      bool isAvailable = await NfcManager.instance.isAvailable();
      print("nfc available");
      NfcManager.instance.startSession(
        onDiscovered: (NfcTag tag) async {
          int? rfid;
          MifareClassic? platformtag = MifareClassic.from(tag);
          if (platformtag == null) {
            print('Tag is not compatible with NDEF');
            return;
          }
          setState(() {
            rfid = int.parse(List.from(platformtag.identifier).join(''));
            if (rfid != null) {
              cards.add(rfid!);
            }
          });
        },
      );
    } catch (e) {
      print('no nfc detected?');
      print(e);
    }
  }

  @override
  void initState() {
    // TODO: implement initState
    super.initState();
    nfcInit();
  }

  @override
  Widget build(BuildContext context) {
    return Scaffold(
      appBar: AppBar(title: Text('test')),
      body: Container(
        color: Colors.blueGrey,
        child: ListView.builder(
          itemCount: cards.length,
          itemBuilder: ((context, index) {
            return Center(
              child: Text(
                cards[index].toString(),
                style: TextStyle(fontSize: 25, color: Colors.white),
              ),
            );
          }),
        ),
      ),
    );
  }
}
```

run this app and test it by reading a card, then lock the screen (while the app is running) and unlock it, if you try to read a card now, then it doesn't read and the sound emitted from the device is different (indicating the session is stopped).

aadarshadhakalg, 2022-12-17:
I couldn't reproduce this issue.

Nairaud, 2024-03-06:
Hi we have the same issue with our app on a Samsung Galaxy A04s the app will detect the tag but the moment we lock the screen and unlock it we can't scan anything on the app anymore, just opening the task manager of the phone and coming back to the app does make the app able to scan tags again but nothing fixes it without putting the app in the background and foreground again.