okadan/flutter-nfc-manager #102: Targeting S+ (version 31 and above)

Hesam-Taleghani, 2022-08-10:
Hi! Thanks for t his great project
I work with nfca tags and my app works fine with android 10 but when I want to run it on android 12, It crashes
Caused by: java.lang.IllegalArgumentException: com..: Targeting S+ (version 31 and above) requires that one of FLAG_IMMUTABLE or FLAG_MUTABLE be specified when creating a PendingIntent.
E/AndroidRuntime(21272): Strongly consider using FLAG_IMMUTABLE, only use FLAG_MUTABLE if some functionality depends on the PendingIntent being mutable, e.g. if it needs to be used with inline replies or bubbles.
E/AndroidRuntime(21272): 	at android.app.PendingIntent.checkFlags(PendingIntent.java:382)
E/AndroidRuntime(21272): 	at android.app.PendingIntent.getActivityAsUser(PendingIntent.java:465)
E/AndroidRuntime(21272): 	at android.app.PendingIntent.getActivity(PendingIntent.java:451)
E/AndroidRuntime(21272): 	at android.app.PendingIntent.getActivity(PendingIntent.java:415)
E/AndroidRuntime(21272): 	at com.example.coollog.MainActivity.onResume(MainActivity.kt:13)
E/AndroidRuntime(21272): 	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1476)
E/AndroidRuntime(21272): 	at android.app.Activity.performResume(Activity.java:8441)
E/AndroidRuntime(21272): 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:5200)

xdamien7, 2022-08-17:
Hello,

It took me a while but this sold the problem for me

1.modify your MainActivity.kt file (search in your project) in the app/src/main/kotlin/com  folder tree

`
package com.yourpackage.name

import android.app.PendingIntent
import android.content.Intent
import android.nfc.NfcAdapter
import io.flutter.embedding.android.FlutterActivity

/// Workaround for disabling the default-tag-detection while the app is foreground.
class MainActivity: FlutterActivity() {
  override fun onResume() {
    super.onResume()
   
    val intent = Intent(context, javaClass).addFlags(Intent.FLAG_ACTIVITY_SINGLE_TOP)
    val pendingIntent = PendingIntent.getActivity(
    context,
    0,
    intent,
    PendingIntent.FLAG_IMMUTABLE or PendingIntent.FLAG_UPDATE_CURRENT // setting the mutability flag 
    )

    NfcAdapter.getDefaultAdapter(context)?.enableForegroundDispatch(this, pendingIntent, null, null)
  }

  override fun onPause() {
    super.onPause()
    NfcAdapter.getDefaultAdapter(context)?.disableForegroundDispatch(this)
  }
}

2. modify your app/build.gradle file

in the bottom of the file  add below dependencies.

below is my copy

dependencies {
   implementation "org.jetbrains.kotlin:kotlin-stdlib-jdk7:$kotlin_version"

    // (Java only)
    implementation("androidx.work:work-runtime:2.7.1")

    // Kotlin + coroutines
    implementation("androidx.work:work-runtime-ktx:2.7.1")
}

Hope this helps (can't guarantee this will work for you, but it may guide you)

Grrravity, 2023-08-16:
Bump this :P

Trying @xdamien7 's proposal but would be nice to have this fixed

ManXliu, 2023-10-20:
有用

LeszekPieczaba, 2024-12-14:
Hi,
I've stumbled upon quite the same error while testing 4.0.0-dev version (https://pub.dev/packages/nfc_manager/versions/4.0.0-dev). This issue shows up when I try to invoke`NfcManagerAndroid.instance.enableForegroundDispatch()`. I've tried:
- to apply the workaround described by @xdamien7,
- `$kotlin_version`: `1.9.0` and `2.0.20` - but I am not sure what version exactly should it be

Could someone point me a direction what I could try in order to get it to work?

@Grrravity how did you manage to get it to work? Meaning to set the mutability flag? Because in my case the workaround doesn't seem to have any effect.

Any help is very much appreciated.

Thank you in advance.

Below is my stack trace:

```
E/flutter ( 8637): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(IllegalArgumentException, java.lang.IllegalArgumentException: com.example.my_app: Targeting S+ (version 31 and above) requires that one of FLAG_IMMUTABLE or FLAG_MUTABLE be specified when creating a PendingIntent.
E/flutter ( 8637): Strongly consider using FLAG_IMMUTABLE, only use FLAG_MUTABLE if some functionality depends on the PendingIntent being mutable, e.g. if it needs to be used with inline replies or bubbles., Cause: null, Stacktrace: java.lang.IllegalArgumentException: com.example.my_app: Targeting S+ (version 31 and above) requires that one of FLAG_IMMUTABLE or FLAG_MUTABLE be specified when creating a PendingIntent.
E/flutter ( 8637): Strongly consider using FLAG_IMMUTABLE, only use FLAG_MUTABLE if some functionality depends on the PendingIntent being mutable, e.g. if it needs to be used with inline replies or bubbles.
E/flutter ( 8637): 	at android.app.PendingIntent.checkPendingIntent(PendingIntent.java:435)
E/flutter ( 8637): 	at android.app.PendingIntent.getActivityAsUser(PendingIntent.java:551)
E/flutter ( 8637): 	at android.app.PendingIntent.getActivity(PendingIntent.java:537)
E/flutter ( 8637): 	at android.app.PendingIntent.getActivity(PendingIntent.java:501)
E/flutter ( 8637): 	at dev.flutter.plugins.nfcmanager.NfcManagerPlugin.nfcAdapterEnableForegroundDispatch(NfcManagerPlugin.kt:91)
E/flutter ( 8637): 	at dev.flutter.plugins.nfcmanager.PigeonHostApi$Companion.setUp$lambda$11$lambda$10(Pigeon.kt:779)
E/flutter ( 8637): 	at dev.flutter.plugins.nfcmanager.PigeonHostApi$Companion.$r8$lambda$5tdKX3SNtkxXgHkK9CQ10Z8ll3E(Unknown Source:0)
E/flutter ( 8637): 	at dev.flutter.plugins.nfcmanager.PigeonHostApi$Companion$$ExternalSyntheticLambda38.onMessage(D8$$SyntheticClass:0)
E/flutter ( 8637): 	at io.flutter.plugin.common.BasicMessageChannel$IncomingMessageHandler.onMessage(BasicMessageChannel.java:261)
E/flutter ( 8637): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:292)
E/flutter ( 8637): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:319)
E/flutter ( 8637): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(D8$$SyntheticClass:0)
E/flutter ( 8637): 	at android.os.Handler.handleCallback(Handler.java:958)
E/flutter ( 8637): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/flutter ( 8637): 	at android.os.Looper.loopOnce(Looper.java:230)
E/flutter ( 8637): 	at android.os.Looper.loop(Looper.java:319)
E/flutter ( 8637): 	at android.app.ActivityThread.main(ActivityThread.java:8919)
E/flutter ( 8637): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter ( 8637): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:578)
E/flutter ( 8637): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1103)
E/flutter ( 8637): , null)
E/flutter ( 8637): #0      PigeonHostApi.nfcAdapterEnableForegroundDispatch (package:nfc_manager/src/nfc_manager_android/pigeon.g.dart:808:7)
E/flutter ( 8637): <asynchronous suspension>
```