okadan/flutter-nfc-manager #108: Open nfc default NFC reader.

Tanvirul-swe, 2022-09-28:
This issue only occurs on Android whenever I try to scan NFC tag using my own app where it triggers the default NFC reader of the phone.

UnbreakCode, 2022-10-04:
you have to priorize your running app in the MainActivity.kt.

https://developer.android.com/reference/kotlin/android/nfc/NfcAdapter#enableForegroundDispatch(android.app.Activity,%20android.app.PendingIntent,%20kotlin.Array,%20kotlin.Array)

73VW, 2022-12-02:
We seem to experience the same issue in one of our apps.

@Tanvirul-swe have you been able to solve this?

73VW, 2022-12-02:
@UnbreakCode I've downvoted because you didn't explain how to use this function.

I'll now vote your second comment as it really provides a way to solve the problem!
Thanks!

batoul-alani, 2023-01-30:
I try to add these lines on MainActivity.kt
but when I run the app it gives me the Error: 
[image]

> idk why to downvote my answer even if it is correct, just take a look at the example: https://github.com/okadan/nfc-manager/blob/a20a2420597ffdc545bb7321db4f0b86003124da/android/app/src/main/kotlin/com/naokiokada/nfcmanager/MainActivity.kt#L10

UnbreakCode, 2023-01-30:
> I try to add these lines on MainActivity.kt but when I run the app it gives me the Error: [image]
> 
> >

try to use PendingIntent.FLAG_IMMUTABLE as flagType

batoul-alani, 2023-01-30:
I am not very familiar with java can you explain how will be the code?

> > I try to add these lines on MainActivity.kt but when I run the app it gives me the Error: [image]
> > >
> 
> try to use PendingIntent.FLAG_IMMUTABLE as flagType

batoul-alani, 2023-01-30:
Thanks I solve it by using this line:
val pendingIntent = PendingIntent.getActivity(context, 0, intent, PendingIntent.FLAG_IMMUTABLE)
> I am not very familiar with java can you explain how will be the code?
> 
> > > I try to add these lines on MainActivity.kt but when I run the app it gives me the Error: [image]
> > > >
> > 
> > 
> > try to use PendingIntent.FLAG_IMMUTABLE as flagType

UnbreakCode, 2023-01-30:
> Thanks I solve it by using this line:
> val pendingIntent = PendingIntent.getActivity(context, 0, intent, PendingIntent.FLAG_IMMUTABLE)

exactly!

batoul-alani, 2023-01-31:
how can we fix the same issue in ios? 
is there any solution for that?

UnbreakCode, 2023-01-31:
> how can we fix the same issue in ios?
> is there any solution for that?

as far as I know it is not possible for iOS due to API limitations. But maybe the background tag reading is a "workaround", I don't know. https://developer.apple.com/documentation/corenfc/adding_support_for_background_tag_reading