okadan/flutter-nfc-manager #109: instance.isAvailable returning false for iOS on a device I know NFC works on

SRadfordM, 2022-10-03:
Working on a country specific version of an application at my company.  I ported over everything from one code base to another and made only minor changes for country specific code.  The NFC reader on this new application does work for Android, but does not work for iOS. I receive no errors...just need some help understanding what possibly could be missing. 

Within the NFC scanner widget, the log I get back in the database shows that isAvailable is returning false and it's not an error being thrown.  

[image]

I have the same version in pubspec.lock on both applications: 

[image]

I have NFC enabled in the app identifier and provisioning profile: 

[image]

[image]

I have the three steps for the entitlements and info.plist copied over from the application where it works.  

[image]

[image]

closed 2022-10-04 by SRadfordM

THJLearn, 2023-01-17:
I have the same problem, my phone is iPhone8 Plus iOS15.4 system, see the status is off, what is the solution?

mahirandigital, 2023-12-19:
> I ge

how you resolve?