okadan/flutter-nfc-manager #131: Unable to read tag from iOS.

vasilycrespo, 2023-03-15:
Hello, first of all just want to thank you for the excellent work done on the plugin!

I am having issues trying to read a simple nfc tag. When I press a button I call this method:

`
  void _nfcRead() async {
    bool isAvailable = await NfcManager.instance.isAvailable();
    print ('start nfc session...');
    print('isAvailable ${isAvailable}');
    NfcManager.instance.startSession(
      onDiscovered: (NfcTag tag) async {
        print('Do something with an NfcTag instance.');
        NfcManager.instance.stopSession();
        print('Tag data record: ${tag.data}');
      },
    );
  }
`

isAvailable returns true, however, I noticed that no iOS NFC popup is shown on the screen as it should. I have already applied the iOS extra steps:

Adding NFC to capabilities

[image]

Adding to `flutter_app/ios/Runner/Info.plist` the required keys:

`	<key>NFCReaderUsageDescription</key>
	<string>Read the NFC card</string>
	<key>com.apple.developer.nfc.readersession.felica.systemcodes</key>
	<array>
		<string>0003</string>
		<string>04D1</string>
		<string>8008</string>
		<string>80DE</string>
		<string>865E</string>
		<string>8592</string>
		<string>8B5D</string>
		<string>8FC1</string>
		<string>FE00</string>
	</array>
	<key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
	<array>
		<string>A000000812010208</string>
		<string>D2760000850101</string>
	</array>
`

Am I missing something? I still actively trying to solve this issue, so if I can figure it out first I let you know. Thanks!

About my workspace:

Mac 12.6.3
Iphone 13PM ios 16.3.1
nfc_manager: ^3.2.0
Flutter 3.7.6 • channel stable • https://github.com/flutter/flutter.git
Framework • revision 12cb4eb7a0 (2 weeks ago) • 2023-03-01 10:29:26 -0800
Engine • revision ada363ee93
Tools • Dart 2.19.3 • DevTools 2.20.1

vasilycrespo, 2023-03-15:
Hi, I already solved the issue. Not sure whats wrong I think it was something on my end. After deleteting the flutter app, restarting both the mac and the phone and clearing the cache on xcode, and rebuilding now the NFC popup is working.

LBYPatrick, 2023-10-18:
Have these copied to ``Info.plist``!

```xml
<key>com.apple.developer.nfc.readersession.felica.systemcodes</key>
	<array>
		<string>0003</string>
		<string>04D1</string>
		<string>8008</string>
		<string>80DE</string>
		<string>865E</string>
		<string>8592</string>
		<string>8B5D</string>
		<string>8FC1</string>
		<string>FE00</string>
	</array>
	<key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
	<array>
		<string>A0000002310100000000000000000000</string>
		<string>A0000002310200000000000000000000</string>
		<string>A0000002480300000000000000000000</string>
		<string>A00000006510</string>
		<string>A0000000651010</string>
		<string>315041592E5359532E4444463031</string>
		<string>325041592E5359532E4444463031</string>
		<string>44464D46412E44466172653234313031</string>
		<string>A00000000101</string>
		<string>A000000003000000</string>
		<string>A00000000300037561</string>
		<string>A00000000305076010</string>
		<string>A0000000031010</string>
		<string>A000000003101001</string>
		<string>A000000003101002</string>
		<string>A0000000032010</string>
		<string>A0000000032020</string>
		<string>A0000000033010</string>
		<string>A0000000034010</string>
		<string>A0000000035010</string>
		<string>A000000003534441</string>
		<string>A0000000035350</string>
		<string>A000000003535041</string>
		<string>A0000000036010</string>
		<string>A0000000036020</string>
		<string>A0000000038002</string>
		<string>A0000000038010</string>
		<string>A0000000039010</string>
		<string>A000000003999910</string>
		<string>A0000000040000</string>
		<string>A00000000401</string>
		<string>A0000000041010</string>
		<string>A00000000410101213</string>
		<string>A00000000410101215</string>
		<string>A0000000041010BB5449435301</string>
		<string>A0000000042010</string>
		<string>A0000000042203</string>
		<string>A0000000043010</string>
		<string>A0000000043060</string>
		<string>A000000004306001</string>
		<string>A0000000044010</string>
		<string>A0000000045010</string>
		<string>A0000000045555</string>
		<string>A0000000046000</string>
		<string>A0000000048002</string>
		<string>A0000000049999</string>
		<string>A0000000050001</string>
		<string>A0000000050002</string>
		<string>A0000000090001FF44FF1289</string>
		<string>A0000000101030</string>
		<string>A00000001800</string>
		<string>A0000000181001</string>
		<string>A000000018434D</string>
[... 197 lines cut ...]
		<string>D27600002200000060</string>
		<string>D276000025</string>
		<string>D27600002545410100</string>
		<string>D27600002545500100</string>
		<string>D27600002547410100</string>
		<string>D276000060</string>
		<string>D276000118</string>
		<string>D2760001180101</string>
		<string>D27600012401</string>
		<string>D276000124010101FFFF000000010000</string>
		<string>D2760001240102000000000000010000</string>
		<string>D27600012402</string>
		<string>D2760001240200010000000000000000</string>
		<string>D4100000011010</string>
		<string>D5280050218002</string>
		<string>D5780000021010</string>
		<string>D7560000010101</string>
		<string>D7560000300101</string>
		<string>E80704007F00070302</string>
		<string>E82881C11702</string>
		<string>E828BD080F</string>
		<string>F0000000030001</string>
	</array>

```