okadan/flutter-nfc-manager #135: Nothing happens when I use this on a Samsung Galaxy A20e

jakusb, 2023-03-27:
Hi, 
I have am testing on a Galaxy A20e, log says NFC is available, but no popup shown.
Any ideas?
The phone can scan NFC outside of app, but does not seem to start session in app.

This code used. 'onTapReadItem continuous' is shown and 'NfcManager.instance.isAvailable[true]' as well.
but 'startSession' never logged.

    loggy.info('onTapReadItem continuous[$continuous}]');
    NfcManager.instance.isAvailable().then((value) => loggy.info('NfcManager.instance.isAvailable[$value]'));
      return NfcManager.instance.startSession(alertMessage: alertMessage,onDiscovered: (NfcTag tag) async {
        loggy.info('startSession');

jakusb, 2023-04-06:
Ok, I think I figured it out. It is a bit confusing that iOS demand to have a popup modal presented to indicate scanning and haptic feedback on a scan.
Android does not seem to require that, and this module does not provide that same experience. 
Might it be an option to facilitate it as an option that can be turned on?

steven11329, 2025-01-10:
I encountered the same issue with the Samsung A54. Did you fix it?