okadan/flutter-nfc-manager #138: Put animation in 'ready to scan' message for iOS

adplant, 2023-04-18:
When you start a NFC session via `NfcManager.instance.startSession()` on iOS there is a pop up that displays and says 'ready to scan'. I've got two questions and/or requests...

1. Is there a way to put an animation (or more plainly, a Widget) in that pop up? (You currently only have the ability to add a `String? alertMessage`).

2. If that isn't possible, does the ability to prevent that pop up from displaying exist? Can it?

Background on request:

When the NFC session starts (via the user clicking a button), I want to provide a visual instruction to them on what to do. I've implemented this on the main screen, but the pop up is blocking it.

adplant, 2023-05-02:
Just checking in to see if anyone has any input on this. Please let me know.

philitell, 2023-08-01:
Is it possible to implement this feature? Is it possible to dismiss the pop-up and display a custom message?

komaxx, 2023-09-06:
As far as I understand NFC scanning on iOS, that's not possible.

For 1:
The scanning sheet is created and shown by iOS when a scanning sessions is started. The API does not allow for arbitrary views, just some text.

For 2:
When actively scanning, there is no way to suppress the scanning sheet.
There is a way to scan for NFC devices in background - but the user flow is different then, and the user has to tap OS-generated notification.

See: [the official documentation](https://developer.apple.com/design/human-interface-guidelines/nfc)