okadan/flutter-nfc-manager #157: How to read all data from mifare ?

KhomsanSaejia, 2023-09-18:
Sorry, I have some problems with when I read NFC. I want to read all data in mifare card (block, sector) in the example code. The response shows the number of blocks and the number of sectors. But there are no card details. Do you have a sample for me?
[image]

conedie, 2023-10-14:
@KhomsanSaejia  Did you find any solution?

conedie, 2023-10-14:
@KhomsanSaejia Researching I found that it is related to the ASCII code.

KhomsanSaejia, 2023-10-14:
@conedie yes I can. For now I can reading mifare classic

jennachennn, 2024-04-25:
I have the same problem. I can only see the quantity of sectors but not the detailed card content. How did you handle this?

KhomsanSaejia, 2024-04-25:
@jennachennn I'm not sure I will check my code again but you can use MifareClassic instead of Ndef

KhomsanSaejia, 2024-04-25:
@jennachennn you can try this.
[image]

jennachennn, 2024-04-29:
@KhomsanSaejia Thank you! I'll give it a try.