okadan/flutter-nfc-manager #163: Crash on old iPhone

ryanhz, 2023-10-26:
Running the example project on iPhone 6,  ios 12.5.7, crash with no stack information, just assembly:

```
dyld`:
    0x100fba410 <+0>:  mov    x16, #0x209
    0x100fba414 <+4>:  svc    #0x80
->  0x100fba418 <+8>:  b.lo   0x100fba430               ; <+32>
    0x100fba41c <+12>: stp    x29, x30, [sp, #-0x10]!
    0x100fba420 <+16>: mov    x29, sp
    0x100fba424 <+20>: bl     0x100fb9850               ; cerror_nocancel
    0x100fba428 <+24>: mov    sp, x29
    0x100fba42c <+28>: ldp    x29, x30, [sp], #0x10
    0x100fba430 <+32>: ret    
```

alastairlundy, 2024-02-20:
@ryanhz I know this issue is from last year but the problem you're facing is that this functionality requires IOS 13 or newer to use the relevant APIs.

ryanhz, 2024-02-20:
> @ryanhz I know this issue is from last year but the problem you're facing is that this functionality requires IOS 13 or newer to use the relevant APIs.

Yes, I was aware of the requirement.  The issue is more in how to gracefully bypass old iPhone with this plugin?

ryanhz, 2024-02-20:
This is an instant crash, The only thing you can see is a flash of the app being opend and closed in the same instant.
I think there is a workaround? as mentioned here:
https://github.com/semlette/nfc_in_flutter/issues/20#issuecomment-570816518