okadan/flutter-nfc-manager #165: NDEF Read bug

Pranet-Hiranandani, 2023-11-21:
Even with using the following code provided on the official documentation for the plugin: 
```
Ndef? ndef = Ndef.from(tag);

if (ndef == null) {
    print('Tag is not compatible with NDEF');
    return;
}
NdefMessage message = await ndef.read();
```

I am getting a null reading on the ndef.from(tag) function, despite the same code working for me previously and the tag being ndef compatible.

Here's the data outputted from the "tag.data" function: `{nfca: {identifier: [98, 173, 223, 85], atqa: [4, 0], maxTransceiveLength: 253, sak: 8, timeout: 618}, mifareclassic: {identifier: [98, 173, 223, 85], blockCount: 64, maxTransceiveLength: 253, sectorCount: 16, size: 1024, timeout: 618, type: 0}, ndefformatable: {identifier: [98, 173, 223, 85]}} `

In this case the ndefformatable value being non-null clearly shows that even the package reads the tag as ndef compatible. I am unable to build the rest of my app due to this issue, building part of it due to the plugin working perfectly with the same card before.

Am on the latest stable version of both the plugin and flutter.

Please help me with fixing this issue so I can continue building my app.

Pranet-Hiranandani, 2023-11-22:
A flutter upgrade fixed the issue. Not sure of the cause of the issue but as of now it is resolved.

Pranet-Hiranandani, 2023-11-22:
The issue is randomly occuring. Not sure of the cause of the issue.

Adithyan2295, 2023-11-28:
> The issue is randomly occuring. Not sure of the cause of the issue.

https://stackoverflow.com/questions/25512204/not-able-to-format-nfc-card-using-ndefformatable

according to this stackoverflow solution a newcard supports NdefFormatable .. so for a new card i dont think Ndef? ndef = Ndef.from(tag); works ..I think we have to use final ndefFormatable = NdefFormatable.from(tag); but this code is also not working and I am also stuck in this same issue

jayjayesh, 2024-02-17:
[image]

You could try these all, check which one is work.
in my case, (NfcA) (IsoDep) works.