okadan/flutter-nfc-manager #167: NDEFformatable returning PlatformException(io_exception, null, null, null)

Adithyan2295, 2023-11-28:
[image]

[image]

I have been trying to write to a new card .. ndef.write() also returns the same error..If the card is not empty I can overwrite. Writing to a new card is the issue for me.

hebaothman, 2024-06-02:
Any news about this issue?

pranil-iottive, 2024-06-13:
I'm also getting the same issue, with the same card sometimes the package gives this error **PlatformException(io_exception, null, null, null)** and after a few attempts, I'm getting a small detail with this message **PlatformException(io_exception, Tag is not ndef, null, null)**

Adithyan2295, 2024-06-13:
@hebaothman @carki001 @pranil-iottive 
The issue I was facing was that ,Some cards have  **NDEF formattable technology** which means when the card is empty the value is basically null.So some devices does not recognize this format.Once we write to the card the technology changes to just **NDEF** which we can overwrite however we like. My solution was to change the card that has by default **NDEF** technology which by default have a empty string instead of null when formatted or while writing for the first time. 

This card fixed the issue for me: NTAG216 NFC Tags Cards. Search for NFC cards NTAG 213,215,216
my understanding is some devices does not accept **NDEF formattable technology** causing the issue when card is empty.

[image]
[image]

philitell, 2024-07-16:
Some tags use ndefformatable - this means that such tags can be formatted to ndef.
nfc-manager fails to do so for some tags. 
**Tags that are working:**
I'm using NTAG 213,215,216 tags. This tags can be formatted from ndefformatable to ndef with nfc-manager.

**Tags that are failing:**
When i'm using Mifare Desfire EV3 tags (which are also ndefformatable) i'm getting PlatformException(io_exception, null, null, null) on `await ndefFormatableTag.format(message);`

hebaothman, 2025-01-05:
@philitell did you find a solution?

hebaothman, 2025-01-05:
> @philitell <https://github.com/philitell> did you find a solution?
>
> No, this needs to be solved by this package

So how you solved this problem?

philitell, 2025-01-05:
I didn't solved it - this is still an issue

hebaothman, 2025-01-05:
OK thanks