okadan/flutter-nfc-manager #171: check NFC hardware support on device

Suheb786, 2023-12-15:
there are two checks that i want to implement -

1- check if user's device have NFC
2- check if user's has disabled or enabled the NFC

Note- available method only provide if the NFC is enable or disabled, what about the first check

okadan, 2024-01-23:
This issue will be resolved in v4.0.0. (#128)

Mkohm, 2025-09-10:
In case the device does not have NFC the available check will throw an error. To check on Android whether it is enabled or not we can use `NfcManagerAndroid.instance.isEnabled();` which was not very well documented :)

okadan, 2025-09-30:
As of v4.1.0, you can use [`NfcManager.instance.checkAvailability()`](https://pub.dev/documentation/nfc_manager/4.1.0/nfc_manager/NfcManager/checkAvailability.html).

closed 2025-09-30 by okadan