okadan/flutter-nfc-manager #172: Cannot Run NFC in IOS

mahirandigital, 2023-12-19:
i got this message 
XPC Error: Error Domain=NSCocoaErrorDomain Code=4099 "The connection to service named com.apple.nfcd.service.corenfc was invalidated from this process." UserInfo={NSDebugDescription=The connection to service named com.apple.nfcd.service.corenfc was invalidated from this process.}

how to solve it?

StephanWalters, 2023-12-21:
im running into the same issue now.

hieumt211299, 2024-01-02:
does anysone solve this problem ?

StephanWalters, 2024-01-02:
> does anysone solve this problem ?

I ended up using https://pub.dev/packages/flutter_nfc_kit

this works for both iOS and Android

ejjat0909, 2024-01-04:
> > does anysone solve this problem ?
> 
> I ended up using https://pub.dev/packages/flutter_nfc_kit
> 
> this works for both iOS and Android

I use this package too, but still get error CoreNFC like this
XPC Error: Error Domain=NSCocoaErrorDomain Code=4099 "The connection to service named com.apple.nfcd.service.corenfc was invalidated from this process." UserInfo={NSDebugDescription=The connection to service named com.apple.nfcd.service.corenfc was invalidated from this process.}

can you show your setup and info.plist file for the nfc part?

ejjat0909, 2024-01-04:
Hi, I just found out the trick or how to solve. 
open in xcode. Under the sections targets. Choose your targets > signing and capabilities > + capability
add "Near Field communication tag reading"
Close xcode. do run `pod install.`
before run, please restart your physical iphone. after it just run either in the xcode or flutter run in the vscode