okadan/flutter-nfc-manager #181: `nfc_manager_ndef` doesn't exist on pub.dev

Dohmanlechx, 2024-02-20:
I need to run the latest dev version (`4.0.0-dev`) because otherwise, I would run into some Android Gradle issues.

When I bumped the package to the latest I learned that the NDEF stuff was extracted to a subpackage (though, why?) named `nfc_manager_ndef`, which was stated here: https://pub.dev/packages/nfc_manager/versions/4.0.0-dev#handling-the-nfctag-instance

But that package is not available on pub.dev. I tried importing it via the GitHub URL, but it complained about how I cannot import a GitHub package that relies on a package from pub.dev, or something like that.

So the short question: is there a planned date for when `nfc_manager_ndef` will be uploaded to pub.dev?

Dohmanlechx, 2024-05-07:
@okadan any update?

okadan, 2024-05-07:
@Dohmanlechx Sorry for the delay. I plan to publish the dev version soon (probably within a month).

closed 2025-09-30 by okadan