okadan/flutter-nfc-manager #184: NFC functionality suddenly stopped working with iso7816, throwing PlatformException

khaled-ET, 2024-03-15:
For over 8 months, the NFC functionality with iso7816 has been operating smoothly. However, it has abruptly ceased to function. Upon testing, an error message is encountered when attempting to send commands to the card. The error message reads as follows:

**PlatformException(2, Missing required entitlement, null, null)**

khaled-ET, 2024-03-18:
It appears that iOS restricts the selection of another Applet ID (AID) if it's not listed in the info.plist file. After configuring the info.plist file to include the necessary AID and entitlements related to NFC and ISO7816, the problem was resolved. NFC functionality is now operating smoothly on iOS devices.

closed 2024-03-18 by khaled-ET

ced1check, 2024-12-04:
I've published my app on App Store, so I suspect I have a valid AID, how do I check if info.plist contains appropriate definition  so that NFC starts working?
I'm getting Missing required entitlement at this time. Thanks.