okadan/flutter-nfc-manager #193: Example project App crashes after launch.

UmairSaqibBhutta, 2024-05-30:
When I run the example project on my android device then app crashes on launch and logs :
`Targeting S+ (version 31 and above) requires that one of FLAG_IMMUTABLE or FLAG_MUTABLE be specified when creating a PendingIntent. Strongly consider using FLAG_IMMUTABLE, only use FLAG_MUTABLE if some functionality depends on the PendingIntent being mutable`

UmairSaqibBhutta, 2024-07-02:
@fabienrousseau @ThomasAunvik @fred441a @FritzMatthaeus any help please?

euphoria3k, 2024-08-04:
you need to update the MainActivity.kt file in the example project. Update line 13 to be the following:

val pendingIntent = PendingIntent.getActivity(this, 0, intent, PendingIntent.FLAG_IMMUTABLE) // Updated line