okadan/flutter-nfc-manager #204: ios  onDiscovered in startSession

mohammmedsedky, 2024-09-04:
when i call startSession method show ios system to scan  tag and scan it successfully as image show but  onDiscovered code not call so why?

note : i run the package example and the same error
[image]

Abdul-rehman0045, 2024-09-26:
I am also getting the same issue and I am testing it in iphone 12 pro and the ios version is 17.7

ced1check, 2024-12-04:
On iPhone 12 too, iOS 17.7.1, onDiscovered is not called, however onError is with Missing required entitlement
Same as reported here: https://github.com/okadan/flutter-nfc-manager/issues/207.
Anyone found a solution to this problem?