okadan/flutter-nfc-manager #212: Out of Date;  Permission Denial

HenrikH96, 2024-11-07:
While reviewing my crash logs, I noticed several instances of a strange exception. I've replaced the Tag ID for privacy reasons, but it's a valid ID. Does anyone have any ideas on what might be causing this? I haven't been able to reproduce the issue myself. It seems to occur only on Android devices.

```
 Fatal Exception: java.lang.SecurityException: Permission Denial: Tag ( ID: XX XX XX XX XX XX XX E0 ) is out of date
       at android.nfc.Tag.getTagService(Tag.java:381)
       at android.nfc.tech.Ndef.canMakeReadOnly(Ndef.java:347)
       at io.flutter.plugins.nfcmanager.TranslatorKt.getTagMap(Translator.kt:783)
       at io.flutter.plugins.nfcmanager.NfcManagerPlugin.handleNfcStartSession$lambda$3$lambda$2(NfcManagerPlugin.java:23)
       at android.os.Handler.handleCallback(Handler.java:958)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loopOnce(Looper.java:222)
       at android.os.Looper.loop(Looper.java:314)
       at android.app.ActivityThread.main(ActivityThread.java:8680)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:565)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1081)
```

rohnsht, 2025-01-29:
Getting the same error recorded in Crashlytics. Is there a way to fix this?

wijnanb, 2025-03-07:
I have been receiving the same error.

For me it happens when Tag was lost during consecutive transceive operations. I was able to recover from it by:

* Waiting for a new onDiscovered(NfcTag tag) when the tag is close enough to the Android phone again.
* Continue with transceive operations on the newly received tag. We use Nfcv.from(tag).

You should not call stopSession() or startSession() . Just waiting for onDiscovered to be called again.

ced1check, 2025-09-03:
I have been receiving the same error since fully migrating to a full flutter app on Android, used to be native + flutter.

Do you know if it crashes the app or is simply reported by crashlytics? I'm unable to reproduce and we've been testing this on many devices for the past 6 months.