okadan/flutter-nfc-manager #224: Deprecated toLowerCase Method in nfc_manager v3.5.0

suryakant-dubey-us, 2025-03-24:
Description
We are encountering the following issue while using nfc_manager v3.5.0 with the latest Kotlin version:

`e: file:///Users/system/.pub-cache/hosted/pub.dev/nfc_manager-3.5.0/android/src/main/kotlin/io/flutter/plugins/nfcmanager/Translator.kt:43:15  
'fun String.toLowerCase(locale: Locale): String' is deprecated. Use lowercase() instead.`

I noticed that this issue has already been resolved and merged into the main branch.

Request
Could you please release version 3.5.1 with this fix so that we can use the package with the latest Kotlin version?

Thanks!

AdamRiedl, 2025-03-25:
Yes, can confirm same issue. Please release version v3.5.1

andonigarcialopez, 2025-03-26:
I'm having the same issue, thanks!

DavidProdinger, 2025-04-07:
As a workaround, you can use it directly from GitHub:

```yaml
  nfc_manager:
    git:
      url: https://github.com/okadan/flutter-nfc-manager.git
      ref: master
      # or the specific commit
      ref: effaff035be1a80b02f72b6422e7be9ba6fa5b42
```