okadan/flutter-nfc-manager #227: [Info]: Deployment version 3.5.1

HeropolisDa2ny, 2025-03-31:
Hello, I would like to know when are you going to deploy your version 3.5.1 because it seems like the patch made on this version for the problem related to Kotlin error (`.pub-cache/hosted/pub.dev/nfc_manager-3.5.0/android/src/main/kotlin/io/flutter/plugins/nfcmanager/Translator.kt:43:15 'fun String.toLowerCase(locale: Locale): String' is deprecated. Use lowercase() instead.`) corrects the blocking bug on the new flutter versions.
So it would be great to have a new version if it is ready instead of being forced to use your package from directly from github :
```
  # ! When version ^3.5.1 >= switch to that version
  nfc_manager:
    git:
      url: https://github.com/okadan/flutter-nfc-manager
```

DavidProdinger, 2025-04-07:
Yes, this issue was fixed in #215

@okadan just needs to publish it to [pub.dev](https://pub.dev/packages/nfc_manager), changelog and version is already adjusted for the release.

closed 2025-08-25 by HeropolisDa2ny

dominikIncibit, 2025-09-22:
same issue here? Any workaround?

philitell, 2026-04-03:
Seems the version has never been released